supermerill/SuperSlicer #166: When slicing the printer instrcuts the printhead multiple times to move outside of the printarea.

Kirinkun, 2020-02-11:
### Version
Release Version 2.2.47

### Operating system type + version
Windows 10, 64bit.
Running inside VMWare15.
Host is Linux Ubuntu 19.10

### 3D printer brand / version + firmware version (if known)
Ender 3, DirectDrive modification which limits the bed to 200x230, running Klipper.

### Behavior
When slicing the provided 3MF file, the slicer seems to slice everything correctly.

However, when inspecting the GCode produced the following issues are found:
- The printhead is instructed multiple times with G1 to move to a negative X position (Lines 16859, 58397)
- The printhead is instructed with G1 to move to a X position outside the bed (bedsize: 200x230)
(Line 46628)

This causes the print to fail, since it can't perform the requested move-actions. 
_Is this a new feature request?_
No.

#### Project File (.3MF) where problem occurs
[MasterSpool_V4_002_Export3_PARTA_HoneyComb.zip](https://github.com/supermerill/Slic3r/files/4188236/MasterSpool_V4_002_Export3_PARTA_HoneyComb.zip)

supermerill, 2020-05-04:
It may have been corrected in the latest release already.

Kirinkun, 2020-05-05:
Will try as soon as the next release is live.

closed 2020-06-03 by supermerill