supermerill/SuperSlicer #168: Slicer++ project errors when opened in PS alpha4

neophyl, 2020-02-20:
### Version
2.2.47

### Operating system type + version
Windows 10 1903

### 3D printer brand / version + firmware version (if known)
Prusa MK3 and Cr10

### Behavior
Not strictly a problem with slicer++ but I had a project I created in slicer++ and it works without issue.  However when I try to load that project into Prusa Slicer 2.2.0 alpha 4 I am getting an error. 
It does however load into 2.1.1 and if I save it from v2.1.1 it will then open in alpha4.

[image]

If you have time I'm hoping you can figure out whats wrong as Its really useful to be able to swap back and forth between the slicers with the same projects.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[CGM_v3f_pouch.zip](https://github.com/supermerill/Slic3r/files/4232518/CGM_v3f_pouch.zip)

neophyl, 2020-03-27:
Just a comment that I created a new project in 2.2.48 (the calibration cube I linked in my other issue about empty layers) and tried opening it in the release version of Prusa Slicer 2.2.0 and I get the same ConfigBase::set_deserialize() failed error.

supermerill, 2020-03-27:
gotcha, will look at it.

supermerill, 2021-03-19:
with the latest version, Superslicer project should be now openable with prusaslicer.
But of course, if you overwrite it with prusaslicer, all of the superslicer settings will be erased.

closed 2021-03-19 by supermerill