supermerill/SuperSlicer #170: Under extrude after import config bundle from PrusaSlicer

aleung, 2020-02-26:
### Version
2.2.47.1

### Operating system type + version
MacOS 10.14.5

### 3D printer brand / version + firmware version (if known)
Home made Kossel Mini, Marlin 1.1.9

### Behavior

The print was fine with PrusaSlicer-2.2.0-alpha2. I exported the config bundle from PrusaSlicer and imported into Slic3r++. Printed the gcode generated by Slic3r++ and it was under extrude. The neighboring filaments didn't touch each other.

[configbundle-20200249.txt](https://github.com/supermerill/Slic3r/files/4257388/configbundle-20200249.txt)

supermerill, 2020-02-26:
how much extruded volume was reported by slic3r++ and prusaslicer?
Also, if it's different, i'll need the project file of both. (the .3mf)

aleung, 2020-02-27:
@supermerill I've uploaded all files and photo above.

The reported used filament had tiny different (~2.4%), but the print looks very different.

> PrusaSlicer
> 
> volume: 7788.19
> Used filament: 1.73m, 4164.57mm3, 5.29g
> 
> 
> slic3r++
> 
> volume: 7788.19
> Used filament: 1.69m, 4064.40mm3, 5.16g

supermerill, 2020-02-28:
There is a problem when you "convert" the config from slic3r++ to prusaslicer, because some % aren't over the same settings.
The width % is over the layer height in prusa slicer (old behaviour that doesn't make sense anymore) and over the first nozzle width in slic3r++
hence why the first lines didn't have the same width.

Also, it seems like both are under-extruded, it's just a bit less visible on the right, but i don't think the slicer is to blame.

aleung, 2020-02-28:
I did some more experimental with Slic3r++. Before these tests I had calibrated the extrusion multiplier.

### Left:

- First layer extrusion width: 0.6mm (not use percentage)
- Base layer height: 0.2mm
- First layer height: 0.35mm

### Right:

- First layer extrusion width: 0.6mm (not use percentage)
- Base layer height: 0.2mm
- First layer height: 0.25mm

[image]

The left side one looks OK, but the right side one is under-extruded. Only different setting is the first layer height.

Both projects in [0228.zip](https://github.com/supermerill/Slic3r/files/4267865/0228.zip)

aleung, 2020-02-29:
I figured out that the under-extrude was caused by the gap between nozzle and print bed at z=0. Re-calibrate solved the problem. But I don't know why at first PrusaSlicer gave a result better than Slic3r++.

closed 2020-02-29 by aleung