supermerill/SuperSlicer #173: Avoid crossing perimeters not working well with very thin features

Lucent-RFTG, 2020-03-09:
### Version
Slic3r++ 2.2.47

### Operating system type + version
Windows 10 & MacOS Catalina

### Behavior
Avoid crossing perimeter doesn't appear to be working when printing thin walled items. I conducting research on components printed with TPU (Ninjaflex) and require a feature that forces the nozzle to travel over already printed areas in order to reduce oozing. The the remaining tools designed to mitigate oozing available in Slic3r and this brilliant ++ version do unfortunately not suffice. Having had a look at the source code it appears as if it neglects by default thin walled structures in order for it to work. Is it possible to further develop this that it behaves similar to the combing tool in Cura?

supermerill, 2020-03-11:
Very time-consuming. I can't do that on my spare time.

Lucent-RFTG, 2020-03-11:
I totally understand. I wrote a post-processor in Matlab that uses the Gcode and replaces all rapid movements that go over non-printed areas and forces them over printed areas. As you can imagine that's quite a tedious and slow method of doing getting around this problem. That being said, it works quite reliably. Unfortunately my C++ are pretty much limited to the very basic hence why I haven't made an attempt of writing a function that plugs into Slic3r directly. I wish I could contribute somehow to this...

supermerill, 2020-03-11:
you can post it here, maybe i can learn it.

Lucent-RFTG, 2020-03-11:
Of course I can. I am currently working on a version especially optimised to work with Slic3r++. I will do that maybe next week or week after next, once i have done that. I will in this case ad some more comments for you. I hope that's Ok.

closed 2021-11-19 by supermerill