supermerill/SuperSlicer #174: Mac version crash on config changed

aleung, 2020-03-12:
### Version
2.2.47.1

### Operating system type + version
MacOS 10.14.5 (details in crash log)

### Behavior
* Load the project
* Open preview panel
* In Print Settings, change skirt loops to 0
* It crashes always at this step
[crash.log](https://github.com/supermerill/Slic3r/files/4324916/crash.log)

#### Project File (.3MF) where problem occurs

[mask-hook.3mf.zip](https://github.com/supermerill/Slic3r/files/4324908/mask-hook.3mf.zip)

aleung, 2020-03-12:
It looks like if I switch to 3D Editor View before changing the setting, it won't crash. But I amn't 100% sure.

supermerill, 2020-03-15:
Seems like a render problem. Maybe the graphic card?
Difficult to investigate for me, as i don't have a mac.

aleung, 2020-03-16:
@supermerill Thanks. I found it occurred on PrusaSlicer as well, so it shouldn't be the fault in your code.

Just in case anybody face the same issue: the workaround is to switch off preview before modify any configuration.

closed 2020-03-16 by aleung