supermerill/SuperSlicer #175: Profiles Incompatible Error + Configuration Wizard on every Program Open

Photogad, 2020-03-18:
### Version
2.2.4

### Operating system type + version
Win64

### 3D printer brand / version + firmware version (if known)
Prusa MK3, latest firmware

### Behavior
Every time I open Slic3r++, it tells me the configuration is not compatible with this version, and I have to run the configuration wizard every time. It goes away until I close the program and open ita gain.

[image]
[image]

supermerill, 2020-03-24:
yeah, i think it's because prusa don't want to use "alpha" settings in not-alpha slicer. I'll have to find where to remove that thing.

Photogad, 2020-03-25:
@supermerill  i think you are right. Also i think they change something on the main fork because i cant import my slicer++ profiles into PrusaSlicer anymore
I posted about it here on their git
https://github.com/prusa3d/PrusaSlicer/issues/3909

Its sad because i like to use both forks.  Their program is less buggy and runs smoother than yours, but yours have a lot more features (except the SD card thing and a few other new things theirs have that yours don't have).

I wish they would adopt your features into theirs but i think they wish theirs to only be compatible with n00b or dumb user who are afraid of too many settings

supermerill, 2020-03-26:
In the .idx files in resources/profiles, there are a "min_version" and "max_version", you can put 1.0 and 12.0 to each one, or just delete the .idx files.

supermerill, 2020-03-27:
it can also happen because the version in the .idx is not the same than the one in the .ini

supermerill, 2020-08-17:
Is this still a thing?

closed 2020-11-15 by supermerill