supermerill/SuperSlicer #176: Feature Request: Exclude area when printing

LaserBee, 2020-03-18:
Here's the idea: If I have a large-ish model, sometimes I only want to print a small section of it, to test some feature (clearance on a moving part, space to insert a nut, &c). Normally what I'd do is either keep flipping it and plane cutting it until it's the size I want, or take it into Meshmixer to cut it in pieces. What would be useful is if we could exclude an area, maybe using the "add shape". We could add a shape, say a rectangle, and position it on top of the area we don't want printed. Use it like the "HOLE" shapes on Tinkercad. 

Just an idea! :)

supermerill, 2020-03-24:
if you want to exclude an area, add a modifier mesh and set the nb of perimeters, %infill, top layers, bottom layers to 0 for it.

It's 5 settings instead of one, but it's already possible that way.

supermerill, 2021-03-19:
there's a way now as long as it's not a hole
see #1000

closed 2021-03-19 by supermerill