supermerill/SuperSlicer #177: Feature Request: compatibility with curvislicer

dot-bob, 2020-03-21:
I have been experimenting with a new project located here called curvislicer: [https://github.com/mfx-inria/curvislicer](https://github.com/mfx-inria/curvislicer) 

This project produces gcode that somewhat follows the contour of a model and is somewhat similar non-planar printing but applied post processing.

It is designed to work with ICESL, and while I like it as an experimental slicer it is somewhat limited and rather primitive when compared to Slic3r. 

Take a look as it may be worthwhile to implement.

supermerill, 2020-03-24:
Where is the code that transform the 3D mesh into (not-planar) 2D polygons?

dot-bob, 2020-03-25:
This project takes standard planar sliced gcode and modifies it to produce slightly curved non-planar paths that follow the contour of the model: [academic paper](https://hal.archives-ouvertes.fr/hal-02120033/document).  

All that should be needed in the slicer is to add a little bit of additional information to the gcode so the post processor knows how the generated paths relate to the 3d model.

supermerill, 2020-03-27:
> All that should be needed in the slicer is to add a little bit of additional information to the gcode so the post processor knows how the generated paths relate to the 3d model.

Seems easy enough. 
For the "o X-10.3 Y-5.7 Z0.0", i already output some information in json format at the beginning of the file if you select "Label objects". I can also add the exact shift.
I can also output it in thsi special format, but how it can work with multiple instances? and multiple objects? Imo, in the current state, it has to be edited manually or via a script.

For the "t 0.2", it's only for the first layer height, or every layer has to have the same height? It's never the case, unless you tune your profile for that. I also added some 3D-moves that have variable height. Also, there is the problem of bridges. Best way imo, is to let the user write it itself, as it's his responsibility.

You're not from this cnrs team?

dot-bob, 2020-03-31:
I am not from the team.  I follow the iceSL project and thought it may be something others might be interested in as well.  

Slic3r++ seems to be geared more towards experimental features/ideas and it might be something worth looking into adding.  It still might be a little too experimental as there are many limitations as you point out.  This may cause more support support requests on your end.

If not perhaps just adding a gcode place holder (something like [offset_mesh_0]) to output the offset of the first model and a checkbox under printer settings (similar to firmware retraction) to output R for retraction and P for prime would suffice and those that want to experiment can create their own printer profile to experiment with.

supermerill, 2020-04-04:
> If not perhaps just adding a gcode place holder (something like [offset_mesh_0]) 

seems easy, will do that for next release

siamak06, 2023-07-18:
I have been using Cuvislicer fro a while now, although it is a command line, I have not been able to see it work or unable to make work within Icesl, nevertheless it is very powerful as far as I can see on my prints.
If I can see how it is used in IceSL it would be even better.

I have tried to build the non-planer version of Slis3r on ubuntu 22.04 LTS to no avail.
There is another non Planer tool path generator Slicer4RTN
https://xyzdims.com/2021/01/27/3d-printing-rotating-tilted-nozzle-option/

Good printing

Siamak