supermerill/SuperSlicer #178: Feature Request: Brim Offset

Photogad, 2020-03-23:
### Problem

When printing PETG on textured sheet (needs more smush onto sheet), brim is very hard to remove from prints.

#### Solution
Add variable brim offset so that the user could specify that the brim is not fully attached to the part. For example, they could add an offset of 0.1mm which should still keep the brim functional but allow it to be detached from part easier.

neophyl, 2020-05-12:
I have a request for this new feature.  One thing I have noticed people complaining about is the elephants foot/brim on the Prusa forum and github.  The default profiles they now have use a 0.2mm inwards elephants foot compensation.  The brim position doesnt adjust with that however.  If you dont actually need that much squish what you get is the brim not connecting to the part at all.
Hence users complaining.  I've made several posts advising them to reduce the default elephants foot to get the brim to re-attach to the part.

With your new brim offset I thought maybe it could be used to counteract that, except that the offset has to be positive away from the part.  It would be handy to allow a negative value to move it closer too.  That way you can have a 0.2 inwards elephants foot and also move the brim in too if needed.

Just a thought :)

supermerill, 2020-05-12:
Seems legit. But i also have to ensure it doesn't go inside any object. Not that straightforward.

supermerill, 2020-06-21:
@neophyl i just check but there is no gap even with big elephant foot in superslicer.

neophyl, 2020-06-21:
I just checked in 2.2.51 with a basic cube.  With brim enabled and with an offset of 0,  and with a -0.2mm elephants foot the brim does not attach to the part which is the same as Prusa Slicer.  
The old logic used to be that the brim would move in with the first layer until they changed it, which has been causing some issues as a brim that doesnt connect to the part is pretty pointless.  
I cant see any difference in Super Slicer but its your decision.

supermerill, 2020-06-21:
Oh, right, it's something i changed just after the release.

neophyl, 2020-07-08:
This looks like its working great now (2.2.52.1).  Its much more logical for the old behaviour of moving the brim relative to the actual printed first layer then applying the new offset.
I've tried all the options, exterior, interior, ears, inside holes with various realistic brim sizes and offsets, combined with both positive and negative first layer elephants foot adjustments.  Fantastic work Merill.

closed 2020-08-24 by supermerill