supermerill/SuperSlicer #179: Donation for the author

AndrewFarley, 2020-03-27:
@supermerill I'm happy to donate some to your continued work on your fork of Slic3r (I saw your 2.0 preliminary release just now).  Please give me a way to contact you privately, so I can send you some money.  And/or update the main README.md to contain info on donations.

Continued from: https://github.com/prusa3d/PrusaSlicer/issues/458

LaserBee, 2020-03-28:
I've asked a couple times if he can set up a Patreon, or even just give a Paypal address. He's put a lot of work into this, and I'd love to buy him a pint!

Kirinkun, 2020-05-05:
+1 I'd be pitching into a patreon or a paypal from while to while.

supermerill, 2020-05-08:
Now there's a sponsor button on the github header.
Thank you for your support!

AndrewFarley, 2020-05-26:
Thanks @supermerill been busy man, just joined as a sponsor.  :)  Cheers!

closed 2020-05-26 by AndrewFarley