supermerill/SuperSlicer #182: Release 2.2.0 beta does not compile on the Mac

AndrewFarley, 2020-03-27:
### Version
2.2.0 beta

### Operating system type + version
Mac OS Catalina 10.15.2

### Behavior
* Unable to compile... error...
```
farley:build$ cmake .. -DCMAKE_PREFIX_PATH="$PWD/../deps/build/destdir/usr/local" -DCMAKE_OSX_DEPLOYMENT_TARGET="10.13" -DSLIC3R_STATIC=1
...
/Users/farley/Projects/slicer/Slic3r-2.2.48/src/libslic3r/PerimeterGenerator.cpp:627:69: error: use of overloaded operator '<<' is ambiguous (with operand types
      'basic_ostream<char, std::__1::char_traits<char> >' and 'Slic3r::ExtrusionLoopRole')
                            std::cout << loop->is_loop() <<" test " << (((ExtrusionLoop*)loop)->loop_role()) <<" & " << ExtrusionLoopRole::elrHole <<"\n";
                            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

supermerill, 2020-03-28:
yeah, i know. As it takes hours to compile, it's a slow process to debug these versions.

supermerill, 2020-03-28:
aaand now boost can't be download anymore from their servers....
and there is a problem in the azure ubuntu repo ...

lbibass, 2020-04-15:
Has there been any update? I still can't download the new build…

closed 2020-05-06 by supermerill