supermerill/SuperSlicer #184: Experimental feature request: double continuous perimeter

liftbag, 2020-03-30:
Some friends of mine have thought up an extrusion strategy to build a closed loop wall with a continuous extrusion action, without interruptions, as happens in vase mode.
Although it is impossible to achieve the aesthetic result of a true spiral-shaped movement, we wondered if continuous action could produce a superior aesthetic result compared to the usual slicing strategies. The aim is to have potted objects with sturdy walls.
We decided to ask it here for the dynamism that characterizes this excellent fork of PrusaSlicer.
If you think it is worth trying to evaluate the results, we are a group that could experience the results.

I attach a screen shot of the continuous transition from one layer to another.

[image]

https://myhub.autodesk360.com/ue298421b/g/shares/SHabee1QT1a327cf2b7afd210163a4a40374?viewState=NoIgbgDAdAjCA0IDeAdEAXAngBwKZoC40ARXAZwEsBzAOzXjQEMyzd1C0AjANgA4BWXN0YATALQAzAMb9%2BYgCwR5AJjGNcMAMxqAnPwi5l3KZv7ydaAL4gAukA

liftbag, 2020-03-31:
Just to give credit to whoever had the idea (yatuin), I am attaching the discussion in which it was born.

https://community.pathio.xyz/t/double-wall-vase/102

supermerill, 2020-04-16:
it's going from "internal perimeters first" to "external periemter first"
in the first layer, the external perimeter is pushing against the inner wall, and not in the second layer. So you may see a great inconsistency in the perceived external perimeter thickness because of that.

It's possible but will take a good amount of time to force this result, and i'm not convinced that the "jump" from the first to second layer will not be much more noticeable than the current impl.

liftbag, 2020-04-17:
> it's going from "internal perimeters first" to "external periemter first"
> in the first layer, the external perimeter is pushing against the inner wall, and not in the second layer. So you may see a great inconsistency in the perceived external perimeter thickness because of that.
> 
I started having your same doubts after opening the feature request.
It is probably best to focus your efforts on something else. If you deem it appropriate, I will close the issue.

supermerill, 2020-05-09:
closing for now. If you have something new for that, you can re-open it.

closed 2020-05-09 by supermerill