supermerill/SuperSlicer #185: looping perimeters makes a single line in air

Coffee0297, 2020-03-31:
### Version
2.2.48

### Operating system type + version
win 10
### 3D printer brand / version + firmware version (if known)
klipper, happens with all

### Behavior
When looping perimeters is on i have a file that makes 1 line in the air, i don't think that is intentional right?

[image]

#### Project File (.3MF) where problem occurs

[DK_Maker_Visir_V0.7 6pc.zip](https://github.com/supermerill/Slic3r/files/4408077/DK_Maker_Visir_V0.7.6pc.zip)

supermerill, 2020-04-04:
where is the line in the air?

Coffee0297, 2020-04-04:
The blue lines? 
[image]

Coffee0297, 2020-04-16:
@supermerill no i have that off. blue is overhang preimeter

-Tonny

supermerill, 2020-04-16:
ok, i'll lokk at this bug.

closed 2020-05-06 by supermerill