supermerill/SuperSlicer #186: Feature request: brim override via modifiers

liftbag, 2020-04-01:
### Version
_next releases (actual 2.2.48.1)_

### Operating system type + version
_Any_

### 3D printer brand / version + firmware version (if known)
_Any_

### Request
When printing multiple objects at the same time, it would be helpful, sometimes, to be able to determine which object should have the brim and which should not,
I was wondering if a simple method to implement it, could be to insert the brim parameters into the modifiers.
In this way, it would be sufficient to load the same object as a modifier and assign the desired brim parameters.

_Is this a new feature request?_
Yes

supermerill, 2020-04-16:
Will look to see if it's possible.

liftbag, 2020-05-09:
Thanks @supermerill. It works with the latest nightly build. Brim ears and brim inside holes don't seem to work, but I probably need to understand how they work. I will investigate.

[image]

[image]

liftbag, 2020-05-09:
@supermerill there is probably an oversight. Skirt and brim settings also appears in the layer height modifier, although it is not accepted.

liftbag, 2020-05-09:
Brim ears works well. I still don't understand how brim inside holes works.

[image]

supermerill, 2020-05-09:
"brim inside holes", when activated allow "external brim" to be lay down even if it's inside a hole (surrounded by something).
If you use the modifiers, you may need to have the exact same settings for the inner object and the outer one (if they are two different objects)

liftbag, 2020-05-11:
Feature request fully integrated for me. I'm going to close.
@supermerill  we have a telegram group on slic3r++ (and PrusaSlicer) and this feature is highly appreciated. Thanks again.

closed 2020-05-11 by liftbag