supermerill/SuperSlicer #188: Thin wall discarded.

rongith, 2020-04-03:
### Version
2.1.44 (perhaps before) to 2.2.48

### Operating system type + version
Win10x64

### 3D printer brand / version + firmware version (if known)
PrusaMK3s

### Behavior
I don't know if that is a feature not well understood or a bug. Some walls are discarded because they are "thin walls". The area is designed with some "selective infill" technique.
With "Overlapping perimeters" and "Thin walls" unticked.
[image]
With "Overlapping perimeters" ticked and "Thin walls" unticked.
[image]
With "Overlapping perimeters" unticked and "Thin walls" ticked.
[image]
With "Overlapping perimeters" ticked and "Thin walls" ticked.
[image]

#### Project File (.3MF) where problem occurs
[03802-r01j - 030-Lint50-I3D.thinwall.3mf.zip](https://github.com/supermerill/Slic3r/files/4430360/03802-r01j.-.030-Lint50-I3D.thinwall.3mf.zip)

supermerill, 2020-04-04:
work as intended.
in 1st case, it's too thin to extrude with perimeters, and you say "i don't want thin walls"

closed 2020-05-02 by supermerill

rongith, 2020-05-02:
Ok Thank you. I wasn't aware that you can even suppress thin walls. 

It is just that I sliced my part with that new version but with settings updated from an older version. I didn't notice the change in handling thin feature. I just updated slic3r, slice and print. But my parts came out differently from the parts I printed with previous version. There were no thin walls on the 9 parts I printed (20-24h print).

Forgot to close, my bad.