supermerill/SuperSlicer #189: Using PLA as support for PETG (and vice-versa) Cannot get wipe tower

TNDavid, 2020-04-05:
Version 2.2.47
I have found that using combination of PLA and PETG, one as support for the other works very well, supports easily removed leaving a good surface, but using Slic3r ++ I cannot get a wipe tower. I set the one for supports as non soluble, since the wipe tower does not support soluble.  I set one extruder for the supports, and it slices that way, but there is no wipe tower, so it just switches filaments and continues printing with no purge.  What am I missing?  Enabling the wipe tower sets the support extruder back to 0

TNDavid, 2020-04-06:
I have been experimenting with settings for hours now, this works in Prusaslicer 2.0.0, with printer setting for "Original Prusa MK3 MMU2" printer setting, and "0.2MM Soluble Interface MK3" print setting, and the support material set as soluble.
In Slic3r++ the same settings will not allow the tower to be enabled. Even though the support material is soluble ( I tried PVA stock material also), I still get the tower does not support non-soluble supports! Somehow the soluble setting is not getting through.
I really want to use Slic3r++, as I cannot print PETG well without the Skinnydip settings!  I may have to resort back to using the Skinnydip add in with Prusaslicer again, but the Slic3r++ implementation of it is much better.

TNDavid, 2020-04-07:
I have started from scratch, but this will not resolve.  Seems to be a bug when setting the support material to soluble, and assigning it to a different extruder.  Still get the error message, and wipe tower cannot be enabled.
[image]

Eboman123, 2020-04-07:
I'm pretty sure you need to go to the Support Material page and change "Z-Offset" to none.  After that, you can choose Soluble Supports and it will let you use the Wipe Tower.  if that doesn't work...let me know.  It was something like that which helped me to get the wipe tower enabled.  Thanks!

TNDavid, 2020-04-08:
> I'm pretty sure you need to go to the Support Material page and change "Z-Offset" to none. After that, you can choose Soluble Supports and it will let you use the Wipe Tower. if that doesn't work...let me know. It was something like that which helped me to get the wipe tower enabled. Thanks!

I had done that by setting both above and below plane to zero, but in multiple testing that reverted back to 0.1.  I went back and set to NONE instead of PLANE, and now it works!  Thanks, you led me to the correct area!  It seemed to slice properly, but it is a very long print, so will try it in the morning.

closed 2020-04-08 by TNDavid