supermerill/SuperSlicer #196: Customizabile automatic arrange distance

guestisp, 2020-04-13:
When automatically arranging multiple objects on the plate, they are kept too far away.
A new parameter to set how many millimeters to use between object would be great.

Most of the time there is no need to use some centimeters, 1mm is even too much :-)

supermerill, 2020-04-16:
Seems easy, the setting seems to be here but is hidden (duplicate_distance)

guestisp, 2020-04-16:
Thank you, i will try manually very soon. Is that value in millimeters?

supermerill, 2020-04-16:
seems like it. But i'm not sure it's within reach of the user. Maybe via command line....

guestisp, 2020-04-16:
It's not in ini config somewhere?

supermerill, 2020-04-16:
it's not added in the list of available settings, so it's not possible to add it as-is i the ui.
I think it's the same for the config file.

closed 2020-05-06 by supermerill