supermerill/SuperSlicer #197: Seam positions not working as expected

ogland, 2020-04-15:
Using recent versions of Slic3r++, seam positions are behaving in an unpredictable way. Corner doesn't hide in corners, aligned doesn't really align as in Prusa Slicer and random seam isn't randomized but rather aligned.

2.2.47.0 and earlier is behaving more like Prusa Slicer. Not sure random is randomized though.

Rendered result:
[image]

2.2.47.0 and earlier is better, but not as expected:
[image]

Prusa Slicer:
[image]

ogland, 2020-04-15:
This is random seam on a different object, same behaviour with only one object:
[image]

supermerill, 2020-04-16:
Random isn't random anymore.

Didn't find any behavior problems on other ones.

"aligned doesn't really align as in Prusa Slicer": what is exactly the problem ?

ogland, 2020-04-16:
It's not aligning in inner corners, so the behavior isn't as easy to design
for as it isn't predictable.

closed 2020-05-06 by supermerill

ogland, 2021-03-16:
Aligned seam is even more far off now in the latest pre-release version. It doesn't prioritize inner corners at all it seems like.
[image]