supermerill/SuperSlicer #199: [FEATURE] Machine Limits on non-marlin FW

bernhardberger, 2020-04-18:
Is there any way you could enable the machine limits settings on non-marlin-fw printer settings? I'm using RRF and Klipper and I'd just like to get better time estimations..

supermerill, 2020-04-18:
It's possible but the rrf and klipper gcode planner isn't implemented into slic3r to deliver perfect estimation, so it will be as they had the marlin estimator, so it may be better but not near as good as for prusa-marlin.

bernhardberger, 2020-04-18:
That makes sense. On my Marlin 2.0 machine I'm almost spot on with estimations, but on my speedy RRF Delta it's off by a big margin..

Is there any way to adjust the 'generic' planner values in some config file? I'm off almost 50% at times. Don't need it to be spot on, but at least in the general ballpark would be nice to get a feeling for some adjustments.

c10l, 2020-05-10:
I agree. Even better would be to only expose the few settings that you can set on Klipper's config rather than the many ones we do for Marlin.

For reference, when I set my g-code to Marlin and set the accelerations to the same values I have in my Klipper config, estimation is usually correct within a 10min error margin for multi-hour prints so I think that's more than good enough.

th33xitus, 2020-05-10:
I have an additional suggestion on that topic.
When using marlin flavor those machine limits are always forcefully applied to the beginning of the gcode. I personally like the function for its very accurate print time estimation, but never liked it to apply those gcodes to the beginning of my files.

I would love an option, like a little checkbox or so, which lets me choose to only use those machine limits for print time estimations or to apply them to the gcode-file as well.
Is this possible? Or is the print time calculation only possible AFTER the gcode has been created and therefore those machine limits must be applied?

supermerill, 2020-05-10:
For gcode-géneration, my goal is to move all these behaviour to a script, a bit like "custom gcode" but with better gui and more powerful language.

closed 2020-06-03 by supermerill