supermerill/SuperSlicer #200: Config Error on Startup

gizzburn, 2020-04-18:
### Version
_Version of Slic3r++ used goes here_
2.2.48
_Use `About->About Slic3r++` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
Prusa Mk3s
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Bondtech BMG using TZB FW
### Behavior
* _Describe the problem_
On starup an error pops up with 
"Preset bundle `Creality` version not found in index: 0.1.0"
I click ok and the programe works ok.

[image]

supermerill, 2020-04-18:
did you download the .48 or the .48.1 that i made to correct this issue?

gizzburn, 2020-04-20:
Hi bud yes I have the .48.1 running with this error.

commit b9ab7641 (2020-04-21): update version
also fix #200
closed 2020-04-21 by supermerill