supermerill/SuperSlicer #201: Slicing issue when using modifier for infill

make92, 2020-04-19:
### Version
2.2.48

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
when adding a modifier for the upper part of a model to change the infill to 100% for example
its loosing all perimeter for the part below the modivier.

[image]

[image]

#### Project File (.3MF) where problem occurs

supermerill, 2020-04-19:
in the meantime, you can add the "perimeter count" in the modifier to make it work.

make92, 2020-04-21:
> perimeter count

ok will do that, thanks a lot

liftbag, 2020-04-26:
[image]

I confirm the issue. I cannot try the #201 nightly build on macOS, because the executable is damaged.

[image]

[image]

supermerill, 2020-04-26:
Fix should be in the 201 build.
You try to execute the .dmg or from the tar archive?

liftbag, 2020-04-27:
> @liftbag Fix should be in the 201 build.
> You try to execute the .dmg or from the tar archive?
> I don't have a mac so i can't test them. I
> even don't know how you use them.

From .dmg. I'll try from the tar.

make92, 2020-04-27:
the latest version fixed the problem for me.
thanks a lot.

closed 2020-04-27 by make92

liftbag, 2020-04-29:
I got to try the release on windows, and I confirm that it works. But the application for macOS, to which the latest release refers, is damaged.

supermerill, 2020-04-29:
I understand, but i can't 'repair' it on my own, as i don't know why/what is "damaged".