supermerill/SuperSlicer #203: Positive value elephants foot can cause perimeter split

neophyl, 2020-04-21:
### Version
2.2.49

### Operating system type + version
Windows 10 64bit

### 3D printer brand / version + firmware version (if known)
Doesn't look printer dependent

### Behaviour
When using a positive elephants foot compensation that is near or above the extrusion width/nozzle size the external and internal perimeters get seperated and the area between gets infill.

[image]

On this model it starts with the internal hole at 0.39 (nozzle size is 0.4mm).  As you increase it it then happens to the external perimeter too.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[positive elephants foot issue.zip](https://github.com/supermerill/Slic3r/files/4512724/positive.elephants.foot.issue.zip)

supermerill, 2020-04-22:
it's problematic?

neophyl, 2020-04-22:
Ok I'll turn off the single perimeter on top option.

The reason its problematic, at least on this model is that it shakes the printer like crazy with the short zig zag movements as it works its way around the button.
As a small part of the Covid effort the local group sewing masks have run out of buttons, which they use on a cloth hand band for the elastic to fit over as extended wear was causing rubbing behind users ears.  I've been printing them off replacements as they cant buy any.  
With your lovely new variable spacing arrangement option I've got them spaced 0.5mm apart so I can fit 100 on my CR10 bed.  You can imagine the shaking with the bed almost completely full  of these.  
Of course the crappy v wheels used on the cr10 dont help :)

supermerill, 2020-04-22:
Just a note, the XY first layer compensation aka elephant foot, should be negative if you want to reduce the effect, contrary to prusa slicer that want you to enter a positive value.
I have done that because it's the way it works for other compensation (negative : shrink, positive: expand)

closed 2020-07-05 by supermerill