supermerill/SuperSlicer #204: Can not turn gap fill off

neophyl, 2020-04-21:
### Version
2.2.49

### Operating system type + version
Windows 10 64 bit 

### 3D printer brand / version + firmware version (if known)
N/A

### Behaviour
Even with Gap fill un-ticked and the gap fill speed set to 0 slicing still produces gap fill

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[Gap_fill_always_on.zip](https://github.com/supermerill/Slic3r/files/4512962/Gap_fill_always_on.zip)

supermerill, 2020-04-22:
that's the gap fill from the solid pattern. Change the solid pattern (infill with expert settings) to rectilinear instead of rectilinear(filled).

neophyl, 2020-04-22:
Merill, even with all 3 infill patterns (top, bottom, solid set to rectilinear and with gap fill unticked and speed set to zer that slot button model shows 3 sections of gap fill on the preview.  I even tried turning off force 100% fill volume but the gapfill is still there.

supermerill, 2020-04-22:
the label for the bottom are reverse (filled vs no-filled)
[Gap_fill_off.zip](https://github.com/supermerill/Slic3r/files/4516815/Gap_fill_off.zip)

neophyl, 2020-04-22:
You know I never even tried that lol.  Well at least that's an easy one.

closed 2020-04-22 by neophyl