supermerill/SuperSlicer #205: Perimeter seam can be broken with Corners and nearest seam placement

neophyl, 2020-04-22:
### Version
2.2.49

### Operating system type + version
Win 10 64 bit

### 3D printer brand / version + firmware version (if known)
N/A

### Behaviour
When using Corners seam placement on this (nearly) single wall tube the perimeters are broken.  The effect can also be seen on Nearest too but only for several layers (it doesn't travel all the way up, just for about 24 layers).

[image]

Changing to a single perimeter stops the effect.  However for most models multiple perimeters are desirable.
Adding a second instance of the model stops the broken perimeters too.

Rear/Aligned/Random placement does not exhibit the behaviour.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[Corners_seam.zip](https://github.com/supermerill/Slic3r/files/4516171/Corners_seam.zip)

liftbag, 2020-04-23:
As workaround, you can drop the external perimeter width to 0.42mm and enable overlapping perimeters.

[image]

supermerill, 2020-04-23:
neophyl  said that a Rear/Aligned/Random seam works.

liftbag, 2020-04-23:
> neophyl said that a Rear/Aligned/Random seam works.
> overlapping perimeters will create a huge over-extrusion.

Yes, only nearest and corners and nearest produce artifacts.
I did some tests to understand what triggered the problem, and I found that reducing the extrusion width, along with enabling overlapping perimeters, the artifacts disappeared, even with nearest and corners seam. But using the two profile changes individually, it doesn't resolve. It actually looks like a bug.

closed 2020-06-03 by supermerill