supermerill/SuperSlicer #206: Strange behaviour with external perimeters (180 degree U-Turn)

RomRider, 2020-04-22:
### Version
2.2.49

### Operating system type + version
MacOS 10.15.3

### 3D printer brand / version + firmware version (if known)
Ender 5 / Klipper

### Behavior
I recently switched over to slic3r++ instead of Cura and started seeing strange layer lines on my prints (didn't happen with Cura). BTW, same speed, flow, retract, etc.. in Slic3r++ and Cura.

I tracked it down to slic3r++ making strange moves on perimeters. Going one way for one layer and then the other way in the middle of the line for the next layer.
Here is a gif of what I mean:
* First layer: going counter clockwise
* Second Layer: After the first perimeter line is drawn, it does a 180 turn and goes clockwise this time.

[image]

Also with the same settings, Prusaslicer doesn't generate the same behavior (seam: aligned for example)

The result: First: Slic3r++ / Second: Cura (stopped the print, it was only for the purpose of this issue)
[image]
[image]

_

#### Project File (.3MF) where problem occurs
[Slic3r++.3mf.zip](https://github.com/supermerill/Slic3r/files/4518836/Slic3r%2B%2B.3mf.zip)

supermerill, 2020-04-22:
The clockwise/counter-clockwise should only happen when the new perimeter has a part in an overhang area. I found that way it can print them more reliably.

This behavior is harming the surface quality? If yes i'll disable it or make it switchable.

RomRider, 2020-04-22:
There are no overhang here, I've attached the 3mf file to the post if you want to have a look.
It's definitely harming (at least for me), the pressure after the 180 turn is probably higher and thus creating those strange lines.
Making it optional would probably help a lot :)

RomRider, 2020-05-15:
Thanks so much :)

closed 2020-06-24 by supermerill