supermerill/SuperSlicer #207: [Request] Alternative to Skirt for nozzle filling prior to print start...

fiveangle, 2020-04-23:
### Problems with current Skirt feature:
* Skirt is most often used to ensure nozzle is filled prior to start of first layer of object and is close to print start point in order to minimize ooze from nozzle affecting the first layer of the print
* Even when the skirt is located just 2mm from the print start, the first point still shows a noticeable mushrooming out, even in the short distance from moving from the skirt to the print start due to pressure relieving within the nozzle that is not fully resolved by retraction
  * This effect is prominent because of the nature of the first layer "squish" necessary for adhesion
* If bed adhesion is very poor (perhaps bed is not as clean as possible, dust, etc) since the skirt is susceptible because it is just a single extrusion width, it is prone to being knocked off or dragged by the print head during this z-hop when moving from the skirt to the print start point
* Having the skirt ring completely surround the print, it can more easily ruin the print since it often separates from the bed with the nozzle in the middle of it, necessitating the user stopping the print and restarting after removing the failed skirt from around the hot nozzle.
* Essentially, other than for use as a draft shield, when enabling Skirt to serve as a nozzle purge, having a skirt that completely surrounds the print is the worst of all worlds

### Suggestion for an alternative to Skirt for nozzle filling prior to print start

My alternative to the Skirt is to add a separate feature that extrudes a fixed extrusion volume that the user wants to be extruded.  The slicer then calculates an extrusion line orthogonal to the start point of the outermost perimeter of the print, then if it hits a bed edge before the requested volume is extruded, turn to follow the bed edge and keep going until the required volume is reached.  Once that path is determined, inverse path for the start of the print so that the end of this nozzle-filling purge line terminates at 90 degrees to exactly the first start point of the print.  This would leave a single extrusion width "tag" of filament coming out of the bottom perimeter of the finished part, which is even more easily removed than a brim.  When multiple islands are present on the build plate, all but the first print start point can instead use a fixed length, say 10 mm or so, that inversely follows the path of coming out orthogonal from the first start point out to where the configured Skirt would be, then follow along that path, up to 10 mm total. This length could also be configured as a preference but is probably unnecessary (or could be in hidden in "Expert" mode).

While it should not replace the current Skirt feature entirely, this alternate method ensures the start of the print is good and cannot encounter a problem do to the z-hop that currently occurs at the start of any print with skirt enabled (and no brim). So many times has a print failed due to the nozzle lifting and taking the skirt with it to ruin the print start, esp. with PETG and a bed that wasn't ideally cleaned or had the appropriate amount of gluestick (too little it won't stick to PEI bed, no glue stick, it won't separate from the PEI after print, so this issue is a constant struggle). This feature also makes more efficient use of the bed area in the case of a crowded build plate.  

This feature could also be implemented as an addition to the standard "Skirt" feature to improve it so that when using standard Skirt (perhaps for draft shield), the bottom layer can still contain the single extrusion width connecting the skirt to the model start point which helps prevent dragging off of the skirt and also prevents the extra blob squish at the print start point.

Below is a pictorial of the suggestion:
```
     ___________
     |         |<= outermost perimeter of example cube print
     |         |________  
     |_________|       | <= Requisite line length to fully prime
                       |    nozzle based on mm^3 vol set in prefs
                       | 
                       ^Print start point along right bed edge
```

This is a new feature request. I would have submitted my request to the PrusaSlicer repo but based on existing discussion, I don't think here is much interest from Prusa Research since this would be a lowest priority request (just like Ironing, which is working great in Slic3r++ BTW - thanks !!!)

supermerill, 2020-04-23:
like the "; intro line" gcode lines prusa put in their profile (in the Start gcode section)?

fiveangle, 2020-04-23:
It seems I've done a horrible job of explaining :(

No, the "intro line" is a the Prusa-specific fat purge line that occurs outside their printable area.  My suggestion has nothing to do with that.  Their feature is supposed to eliminate the need for a skirt since their "intro line" does fill the nozzle, but then it still has to move over to the print start point, and when it does, the ooze that occurred during that short time smashes down on the bed and shows an obvious blob.  It isn't big, but it can be annoying, and sometimes it still doesn't work right.  For example, sometimes at the print start, if the squish is low (in an effort to not get that curling wave that occurs when printing large solid areas) the start can curl up onto the nozzle for the first couple of CM or so, before settling down.  The ability to have the print start outside of the model area, then connect right into it ensures the model is always "good", even at the start point.

supermerill, 2020-04-23:
understood

fiveangle, 2020-04-23:
I mentioned that:
> When multiple islands are present on the build plate, all but the first print start point can instead use a fixed length, say 10 mm or so, that inversely follows the path of coming out orthogonal [...]

When I say "...where the configured Skirt would be..." I mean *if* a skirt had been enabled (which it would not be in this scenario).  Basically, move to a point 10mm of extrusion along the path orthogonal to the outer perimeter start, then turning onto the path of the skirt (if it were configured) until the required length is obtained, then inverse that path for the start of that perimeter.

Naturally this feature implies "outer perimeter first" since it is required in order to function.

supermerill, 2020-04-23:
the skirt is printed alone before the first object, unless you select "complete individual objects".
Also sometimes there are an object A between object B and the skirt.
[no_path_to_object.zip](https://github.com/supermerill/Slic3r/files/4524366/no_path_to_object.zip)

It may be moderately difficult to replace the move from the skirt to the first object, but everything after that is very difficult.

fiveangle, 2020-04-23:
Moving between islands by extrusions won't work either unless you come out of the outer perimeter immediately adjacent to the entry extrusion, and then you would have to create a non-colliding path around the bed over to the next island.  This is making things too complicated.

> It may be moderately difficult to replace the move from the skirt to the first object, but everything after that is very difficult.

One implementation would be simply enforcing the "rear" seam position then replacing the start point a CM "rear" of that point, or bed edge if hit first.

But, yes, to get the feature to work ideally and in all cases, it seems like it will be too much effort.  Thanks for the discussion on it !