supermerill/SuperSlicer #209: Config and config bundle export is broken

liftbag, 2020-04-25:
### Version
_2.2.49_

### Operating system type + version
_macOS Catalina and win10 Pro build 10240_

### 3D printer brand / version + firmware version (if known)
_Any. Tested also with fresh install for Prusa i3 MK3S_

### Behavior
* _The config and config bundle export seems broken_
* _Steps needed to reproduce the problem_
  * _Open the slicer_
  * _Choose File > Export > Export config... or Export config bundle..._
* _Expected Results_
  * _config.ini or config_bundle.ini should be generated_
* _Actual Results_
  * _a warning window opens with the following error_

[image]

The problem also occurs after the total deletion of the configuration folder, and resetting to the default settings for a Prusa FFF printer.

_Is this a new feature request?_
No

sonictrain, 2020-04-25:
Same problem on Linux aswell.

ashinhood, 2020-05-05:
Same problem on Windows x64 version

liftbag, 2020-05-07:
Problem solved with the #205 nightly build. Thanks @supermerill.

closed 2020-05-07 by liftbag