supermerill/SuperSlicer #210: Skirt extrusion width

stevenignys, 2020-04-25:
### Version
Slic3r++ 2.2.49

### Operating system type + version
Windows 10 (build 18362), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Raise3D Pro2

### Behavior
Using a 0.8mm nozzle and with extrusion widths set to 110% the skirt still comes out as a 0.4 (possibly) extrusion width so massively under extrudes and dosn't stick
[image]

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
not allowed to upload .3MF files?

stevenignys, 2020-04-25:
This happens when extruder 2 is 0.8mm and extruder 1 is 0.4mm. I select the part to use extruder 2 but the skirt does no change. If I manually go into the multiextruder page and set support material/ raft / skirt extruder to 2 it is then correct

supermerill, 2020-04-26:
ok, i understand why this bug occur. Shouldn't be too hard to correct.

closed 2020-06-03 by supermerill