supermerill/SuperSlicer #211: Auto cooling: max speed reduction in %

guestisp, 2020-04-27:
In autocooling option, filament section, please add the feature to reduce the printing speed up to a defined % (based on the feature currently being printed) like with Simplify.

In example, if we have different speeds for inner perimeter, external perimeters and infill, if you have to slow down, you could slow down up to a defined % (in example 20%, so 60-20% = not less than 48mms)

You could use both: minimum printing speed set to a value in mm/s and a slow down %, you keep the higher one of the two.

supermerill, 2020-04-28:
Seems legit and not hard.

guestisp, 2020-04-28:
You are the best developer in the world.

supermerill, 2020-04-30:
I didn't read that well the first time. i was thinking about moving the min from mm to %. But that's not the request.

Currently, the slowdown that happens when an extruder do its job too quickly:
 - reduce the fastest speed to a below threshold.
 - dont reduce the speed below the min.

but before, the speed was reduced by a percentage. It was changed because that creates problems (too slow).

Also, there is the question about the external perimeters.

The old code is here, and there are a switch but it's hidden.
The easy way is to put this switch as a setting.
Maybe adding the min_print_speed to the old algo.

or maybe i still didn't understand the problem?

guestisp, 2020-04-30:
Yes , moving the min from mm/s to a % or adding a new value to keep both (you will use the higher computed value of the two)

Look at S3D, it has a slow down in %

This because if i'm printing at 100mms (because i have a Volcano nozzle or similiar) but my preset is configured with a min speed set to 20mms, the slow down would be too much and lead to artefacts. With a % value, i can set that to 20% and slow down would be up to 80mm/s.

Much easier if you change nozzle or printing parameters or fan blower but keeping the same filament unchanged (and thus the same auto cooling)

Is this something stupid? Im just thinking, to me, a percentage looks more correct

guestisp, 2020-04-30:
I think two values are needed : min speed in mm/s (the current setting) and max speed reduction in %

You slow down not less than min speed and not more than the %

If you print at 100mms, max reduction set to 20%, min speed set to 30mm/s, if you have to slow down a lot, let's say to 50mm/s you will be blocked by max speed reduction (100-20%=80)

supermerill, 2020-06-03:
I've just done a pass on this algorithm, reworking the explanation text.
I'm going to improve it for the next release.

guestisp, 2020-06-25:
20% should mean that you can slow down not more than 20%. In example, if speed i 100mm/s, you can't go lower than 80mm/s

It's a reduction, not a "max speed".

supermerill, 2020-06-25:
I think that's what I have done.

closed 2020-08-24 by supermerill