supermerill/SuperSlicer #212: [Request] More accurate preview key/scale

Photogad, 2020-04-29:
This has been this way in Slic3r forever, I'm not sure if it can be fixed, but just asking in case it can.

The preview scale (key) in the upper left corner not being accurate has always bothered me.

In below example, I'm looking at layer height. It's same 0.20mm layer height for all the layers.

So why does the preview scale go from 0.000 to 0.390? Lol. :) It doesn't even show the exact color for the only layer height there (0.200).

(this applies to all the preview modes, not just layer height)

[image]

liftbag, 2020-04-29:
Mmm, 0.390 is 0.4 (the nozzle size) X 0.975. If you set a 0.6 nozzle the scale is adjusted from 0 to 0.585 that is 0.6 x 0.975 again.
The color is exact to me, because it turns slightly towards green, being the pure yellow used for 0.217.

Photogad, 2020-04-29:
> Mmm, 0.390 is 0.4 (the nozzle size) X 0.975. If you set a 0.6 nozzle the scale is adjusted from 0 to 0.585 that is 0.6 x 0.975 again.
> The color is exact to me, because it turns slightly towards green, being the pure yellow used for 0.217.

See thats exactly the shit im talking about, way too confusing haha

supermerill, 2020-04-29:
what should be the right behaviour? everything in sand-red?

Photogad, 2020-04-29:
Here are some example.

### 1 - In this example, it is correct how I would like all of the preview to be. We have items specifically on the preview key with their own color and clearly defined. I know these are named items and not numbers, but I am just using this as an example for items to follow.

### 2 - In this preview of layer height, the preview key have many colors that don't exist in the print. There are only TWO layer height used in the whole print. And none of the actual layer heights are shown on the key!! I am colorblind and looking at this, I would say that the yellow color is 0.217mm layers, but in fact I know it's 0.200mm layers!! The scale lies.

### 3 - Again, with line width. There are only two line widths used, but the whole scale and colors are shown. And again, I know the darker color is supposed to be 0.450 line width, but there is no 0.450 on the key! The closest number on the key is 0.454mm, but this is wrong and misleading. It lies.

### 4 - Fan speed. Only two fan speeds are used for this print. All the other colors and numbers on the scale key are irrelevant. At least with this one, the numbers on the scale key are correct... there is a 0 and 100. I know the two fan speeds are 0% and 100%.

### 5 - This is perfect!! Only one extruder is used here, only one color, and only one color on the scale.

[image]
[image]
[image]
[image]
[image]

Photogad, 2020-04-29:
`Even if it's not possible to get rid of all the extra unused items or colors on the scale key in upper left corner, can at least it be made so the real numbers are on there??`

Photogad, 2020-04-29:
also posted to
https://github.com/prusa3d/PrusaSlicer/issues/4148

closed 2021-03-19 by supermerill