supermerill/SuperSlicer #214: [Bug] Slic3r++ Takes very long time to generate Gcode

NoxIX, 2020-04-30:
### Version
2.2.49

### Operating system type + version
Windows 10 Pro - 1909
AMD 3600 Cpu

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro - Marlion 2.X.X Bugfix

### Behavior
Problem: Slic3r++ Takes an extended period to generate Gcode.
Steps to reproduce: Slice an object. 
[image]

Model slices quickly in PrusaSlic3r. 

It took so long I thought it would not finish slicing. It will indeed finish.

#### Project File (.3MF) where problem occurs

https://send.firefox.com/download/8e4ca760659aa2ca/#Nl2lq2ZM8x08m3WRWdvyew

supermerill, 2020-04-30:
I've uncheck "avoid crossing perimeter" (the one that say "slow down g-code generation") and it takes me 3 min on my laptop

for a 11 000 000 polygon object, it's not bad.

supermerill, 2020-04-30:
Sorry, i'm maybe a bit harsh, but i can't do anything with that amount of information.
You can remove a '0' from "model rounding precision" (in slicing tab) if you want it to slice in less than a minute.

closed 2020-04-30 by supermerill

NoxIX, 2020-04-30:
> Sorry, i'm maybe a bit harsh, but i can't do anything with that amount of information. You don't provide the .3mf from prusaslicer, and i don't have a clue about the time it takes you to slices this big model.
> 
> note: You can remove a '0' from "model rounding precision" (in slicing tab) if you want it to slice in less than a minute.

Hey sorry, it wasn't requested or I would have. I would suggest asking for it before closing...

However, I think your don't avoid perimeter might be the issue. *However*, that doesn't explain why it shows a fully sliced model, with the export G-code button, but it claims it's still generating it. Why would that be the case?

supermerill, 2020-05-01:
there is the slicing and then the gcode generation. a preview is showed when the first step is finished, but you have to wait for gcode generation before exporting it.