supermerill/SuperSlicer #216: macOS face mask edition (#201) appears damaged

liftbag, 2020-04-30:
### Version
_2.2.49  face mask edition_

### Operating system type + version
_macOS Catalina 10.15.4_
### Behavior
* _The macOS version linked on the release page appears to be damaged and does not work_

[image]

supermerill, 2020-04-30:
I modified the text in the release page.
Until someone with a mac can compile or tell me why it doesn't work, i don't know what to do.

lincolnep, 2020-05-01:
> I modified the text in the release page.
> Until someone with a mac can compile or tell me why it doesn't work, i don't know what to do.

I have a mac and should have enough skills to compile it will have a look at it and let you know how i Go

georgebashi, 2020-05-02:
The 2.2.49 release dmg binary works fine for me on High Sierra (10.13.6). Maybe it's something to do with the notarization requirements on newer macOS releases?

liftbag, 2020-05-02:
Well, the latest nightly build #215 bridge_speed always available works fine on macOS Catalina.
Can I close the issue?

supermerill, 2020-05-02:
if the issue resolve itself, sure.

closed 2020-05-03 by liftbag