supermerill/SuperSlicer #217: [Question Only] Does Slic3r++ Elephant Foot compensation differ from Prusa Slicer?

Photogad, 2020-04-30:
Hi, in Prusa Slicer v2.2.0 alpha 1
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.2.0-alpha1
They added "Adaptive Elephant Compensation"

I was wondering if your fork also uses adaptive mode? 

Because, in your fork, you specify negative values, but in Prusa Slicer, they specify positive values. So I'm not sure this is using the new adaptive mode.

supermerill, 2020-04-30:
i'm using their Adaptive Elephant Compensation, at least if the "sum" of all XY is negative for the perimeter. If it's positive, it just expand as normal.

Photogad, 2020-05-02:
> 
> 
> i'm using their Adaptive Elephant Compensation, at least if the "sum" of all XY is negative for the perimeter (to be clear, if elephant foot is negative and it's still negative if i add all others, then i use the adaptative elephant foot, for the elephant foot value part that is negative). If it's positive, it just expands as normal.
> 
> For calling the prusa function, i multiply by -1 to always have a positive one.

i guess i don't even fully understand how adaptive works, but i just heard from Prusa that their new adaptive mode is better.

supermerill, 2020-05-02:
It avoids empty first layers, for what i know.

closed 2020-05-09 by supermerill