supermerill/SuperSlicer #221: Bad slicing , empty layers

guestisp, 2020-05-02:
### Version
Slic3r++ 2.2.49

S++ is unable to properly slice this 3mf :

https://www.prusaprinters.org/prints/29444-band-head-band-for-face-shield/files

the issue is related to the stl file. Even by generating a new stl file from the openscad, s++ is unable to slice it properly.

PrusaSlicer works correctly with or without the 3mf

guestisp, 2020-05-02:
Also, prusaslicer, with the same 3mf, is much faster, 4-5-6 times faster to build the gcode. S++ takes 5 times more then stop with an error:

```
Empty layers detected, the output would not be printable.
Object name: Head band.stl
Print z: 0.550000
This is usually caused by negligibly small extrusions or by a faulty model. Try to repair the model or change its orientation on the bed.
```

Do you see the broken layer in this image ?
[image]

supermerill, 2020-05-02:
To speed it up, you can deactivate "extra perimeters: filling horizontal gaps on slopes".
It's most of the time unnecessary, and it's slow.

guestisp, 2020-05-02:
Ok, much faster, but still doesn't work properly with this 3mf or even with a newly generated stl from the scad file

supermerill, 2020-05-02:
yeah, it's a bug. will investigate.

supermerill, 2020-05-03:
Ok, i understand why.
In the meantime, you can slice it with "overlapping perimeters".
It's a little piece of software that go bunker because it thinks that's it's quicker/better to print nothing than all that band.

guestisp, 2020-05-03:
Oh, well, usually printing nothing is always quicker than printing something. :-D

guestisp, 2020-05-03:
Side question : try to preview the gcode, do you know why the circle is printed in half and not at the same time? It start to print a leg, then goes ti the right, it came back and then the circle is finished

In vase mode the Path is better

guestisp, 2020-05-03:
With that option turned on, slicing seems ok.

supermerill, 2020-05-04:
I wasn't even that.
It's just that the person that created this profile has done it with "overlapping perimeter" in mind (as it's the way it's done without thin walls in prusaslicer.
But it put 1mm thick extrusions for the external perimeters! As the test in"vertical shell" say, this thickness allow only 1.96 or ticker walls. But the first layer are smaller due to the chamfer!
So there are two ways to slice it:
 - use overlapping perimeters
 - decrease the external perimeter & perimeter thickness to 0.8

And i don't see anything to improve.

>> do you know why the circle is printed in half and not at the same time?

It's because the half-circles are just a little bit thick enough to be printed.

guestisp, 2020-05-06:
Wait, it not a profile issue. I had the same issue by importing the STL file in S++, like I do every time.
I don't use PS anymore except when trying to figure out something strange in S++ and, usually, i start from scratch from an stl, not from 3mf.

supermerill, 2020-05-07:
I can't reproduce the issue, i open the stl with my default settings and everything is ok.
Please paste your project file that can reproduce the issue.

guestisp, 2020-05-10:
Already resolved.

guestisp, 2020-05-11:
This is a 2.1mm thick STL, but sliced it's only 0.6mm
1mm extrusion width, overlapping perimeters set to true

[Band.zip](https://github.com/supermerill/Slic3r/files/4608469/Band.zip)

supermerill, 2020-05-11:
it's a bad stl. Use the repair option in the file menu.

guestisp, 2020-05-11:
Bad stl ? You know why ? It's generated from OpenSCAD

supermerill, 2020-05-11:
and the algorithms made by prusa say that it's not manifold and I can slice it after a go into the windows repair tool....

what can i do? repair openscad?

note that i'm working on an alternative, not sure it's better than openscad, although. https://github.com/supermerill/Pyscad

guestisp, 2020-05-11:
> what can i do? repair openscad?

Obviously not, i'm just asking if you know what is broken so that I can change the scad code

> (btw, it's made with the lastest nightly? or the very old version?)
2019.01.30.nightly (git f914a57) 

I'll try with the latest "stable" from the AppImage

supermerill, 2020-05-11:
I don't know how to give you this information, i left the stl repair to prusa, haven't touch it.

guestisp, 2020-05-11:
With the latest openscad, slicing was ok.

closed 2021-03-19 by supermerill