supermerill/SuperSlicer #222: Thumbnails not added to Gcode

Jack77777777, 2020-05-03:
### Version
Version: 2.2.49
Build: slic3r++_2.2.49_2020-04-21

### Operating system type + version
Windows 10 Pro, Version 1909 

### 3D printer brand / version + firmware version (if known)
Anycubic i3 Mega s, Marlin 1.1.9 Custom Firmware

### Behavior
* _Describe the problem_
No thumbnails added to Gcode. It´s added to Gcode using Prusaslicer
* _Steps needed to reproduce the problem_
Go to Help->Show configuration folder->printer and add **thumbnails = 16x16,220x124** to <yourprinter>.ini
* _Expected Results_
Generating Gcode with thumbnail section in it
* _Actual Results_
No thumbnails generated
Adding Gcode of S++ and Prusaslicer how it should look like.

_Is this a new feature request?_
Perhaps

#### Project File (.3MF) where problem occurs
[All_Files.zip](https://github.com/supermerill/Slic3r/files/4569627/All_Files.zip)

vertigo235, 2020-05-08:
Ahh good to know, I thought I was just doing something wrong

Jack77777777, 2020-05-11:
Great, it´s working !

jneilliii, 2020-05-14:
Was this added as a setting that could be controlled via the UI or only in the ini file directly? I ask because there is a request on the official PrusaSlicer repo for this and it seems you guys are more open to change/making that happen.

3d-gussner, 2020-05-15:
See https://github.com/prusa3d/PrusaSlicer/issues/4127

supermerill, 2020-08-17:
The setting was hidden. I added it into the general tab of the printer.
prusa mini is 16x16,220x124
sl1 is 400x400,800x480
if you set it to 0x0 0x0, you won't export a thumbnail in the gcode. If it's set, you will.

@jneilliii @3d-gussner does this resolve the issue?

jneilliii, 2020-08-17:
That works for me assuming it's available for all printers and not just the mini and sl1. Looking forward to that release.

fernandex00, 2020-08-26:
i got crash when trying to put a value, windows 10 superslicer 2.2.53

jneilliii, 2020-08-26:
I had a similar issue @fernandex00 but got past it by creating a new printer profile.

fernandex00, 2020-08-26:
thank you, i will try that

neophyl, 2020-08-26:
I saw that too, it seems it is initially set to  0 x 0.00, when I tried to edit it caused superslicer to shut down.  I reopened though and for some reason they were now set at 0 x 0 so didnt cause a problem and I didnt have to resave the profile.

supermerill, 2020-08-26:
@fernandex00 @jneilliii
Can you send me a .3mf before you modify the setting that create a crash ?

neophyl, 2020-08-26:
I can reproduce it but not with a 3mf file.

The sequence I used is as follows-

Install clean , no existing profiles etc.
Can run wizard to install printers etc or skip, doesn't matter but make sure you have expert mode on so you can see the printer settings.
Import Config bundle
[PrusaSlicer_config_bundle_work_20190627.zip](https://github.com/supermerill/SuperSlicer/files/5129164/PrusaSlicer_config_bundle_work_20190627.zip)  - theres a hodge podge of 101 presets in there that I need to clean up lol.
That will usually then select one of the imported profiles.
Change to the Printer>general page.  At this point the thumbnail settings look like this -
[image]
Notice the -0.00 for Y on both.
Edit the small Y settings to make 0.  Then try and edit the large Y settings.  For me it then hangs and after a few seconds shuts down.
Once you reopen the software though and go back to that screen the settings are both 0 already and there doesnt seem to be any other issue.  With that printer profile or any of the others.  If I save the 3mf file with the settings as -0.00 and then restart and load the 3mf the values are 0.

I suspect its an initialisation issue.

supermerill, 2020-08-26:
Can't reproduce with your steps, using the local settings to have a clean "install" and a fresh downloaded zip release file.
i have 0 ,0, not 0 -0.00
I'm a bit confused,

randellhodges, 2020-09-01:
I also had the crash with 2.2.53 on windows. I opened up my machine's .ini file and noticed:
`thumbnails = `
I changed it to:
`thumbnails = 16x16,220x220`

Then when I re-opened superslicer I was able to change it to 16x16 and 220x124 without it crashing. I verified it wrote
`thumbnails = 16x16,220x124`
to the machine's .ini file as expected.

supermerill, 2020-09-06:
`thumbnails = `

ok, i can replicate that.
I've added another safeguard to correct this problem.

closed 2020-10-22 by supermerill