supermerill/SuperSlicer #223: Some GUI elements are hidden on a laptop screen

DeviousPenguin, 2020-05-03:
### Version
Version: 2.2.49
Build: slic3r++_2.2.49_2020-05-01

### Operating system type + version
Operating System: Macintosh
System Architecture: 64 bit
System Version: macOS Catalina Version 10.15.4 (Build 19E287)

### 3D printer brand / version + firmware version (if known)
Longer LK4 Pro running Klipper

### Behavior
Preview Mode has GUI elements: 'Travel', 'retractions', '?' that are off the edge of the screen, and unable to use them, here's a screenshot:

[image]

This may be because I'm on a laptop with a smaller screen, but would it be able to make the 'show: extrusion type' bit of the GUI a little thinner so the rest of the GUI can be seen?

#### Project File (.3MF) where problem occurs
This happens with each and every file 

Thanks

supermerill, 2020-05-03:
It's complicated and will takes days to do that correctly.
I think you may have more luck if you test it with prusaslicer and ask for that in the prusa branch.
I merge against prusaslicer at every release.

supermerill, 2020-05-03:
or maybe ir's because i enlarged the combo box (extrusion type here) ... i'll look at make it small on small screens

DeviousPenguin, 2020-05-03:
Thanks, I've downloaded PrusaSlicer and it does look like it has a smaller combo box:

[image]

However I'm still unable to see all the options, so I'm not really sure what the solution is here, maybe move the tickboxes onto the row below or the sidebar? 

Anyway I'll try and request some sort of fix in the prusa branch, thanks. :)

supermerill, 2020-05-03:
i reopened it to let me not forgot to reduce the combo size

supermerill, 2020-05-11:
Wasn't that hard. But no translation for cropped labels.

DeviousPenguin, 2020-05-14:
Thank you very much, I've just updated to the latest build and looks great now, even on my tiny screen! :)

[image]

closed 2020-05-14 by DeviousPenguin