supermerill/SuperSlicer #224: Single skirt in multiple objects

guestisp, 2020-05-03:
When printing multiple objects one by one, a new skirt is added around each object.

This is a waste of time, filament and printing surfare

I propose to add an option to print just one skirt like when printing everything at once

supermerill, 2020-05-03:
#51

guestisp, 2020-05-03:
Sorry for the noise.

supermerill, 2020-05-03:
np. But if you close this, you should reopen the other one with this message.
I'm wondering where i should add this setting... near "complete object individually"?

guestisp, 2020-05-03:
didn't see that the other is closed. I need a vacation.
Keep this open and yes, I think the best place would be near "complete object individually" because if you add something like "Use a single skirt for all object" in the skirt section, it could be misleading, like: using a single skirt globally even when printing multiple object at the same time (the default)

neophyl, 2020-05-12:
I couldn't find this option in the 2.2.50 release, in either the skirt & brim section or the output options.  Am I just missing it ?  I tried it with 2 calibration cubes set to print individually and it was drawing separate skirts around both.

supermerill, 2020-05-12:
should be just below "complete individual objects"
it's called "allow only one skirt loop".

If you have a custom print.ui, obviously, you have to add the line, or modify the configs manually.

neophyl, 2020-05-12:
As far as I know I don't have a custom ui, I wouldn't know how to tell though lol.  I would I remove or tell if I had one ?  Running on Win 10.

btw what I see is 
[image]

Which now Im looking at the screen grab theres a strange dotted artifact to the right of the Complete individual objects checkbox.

supermerill, 2020-05-12:
duh, i forgot to copy the resource directory after the last pull....
 you have to add 
`setting:complete_objects_one_skirt`
at the line 263 of print.ui in resources/print?ui

or download a nighlty build in the ACTION menu.

neophyl, 2020-05-12:
Lol, yeah adding that did it.

liftbag, 2020-05-14:
I wonder if it is more convenient, and probably correct, to move only the _Extruder clearance (mm)_ data in the Printer Settings section, leaving _Complete individual objects_ and _Allow only one skirt loop_ in the current place. After all they describe the machine geometry and it is not necessary to repeat them for each print settings.

supermerill, 2020-05-14:
You're right, it's even an extruder property (if tool-changing).
And you should be able to print y-ordered object, whatever the z clearance is, if you have a y-bar.
but all that is shelf for now, i have other things to do.

I may move it in the printer tab, i'll may wait a bit to add something to it to make it worthwhile.

closed 2020-06-03 by supermerill