supermerill/SuperSlicer #225: Linux/Mac build as release assets

guestisp, 2020-05-04:
For further releases, please add the the linux/mac build as release assets, not as artefacts.

Artefacts needs a proper github account to be accessed and i'm building an automated script to create a working appimage from your Linux build. If using artefacts, the scripts need a github auth (with token or similiar) and thus can't be automated for everyone.

supermerill, 2020-05-04:
Is there something to do it automatically?

guestisp, 2020-05-04:
I think so, but never tried

guestisp, 2020-05-11:
>  i'm building an automated script to create a working appimage

the appimage is build from scratch directly from your source release.

https://gist.github.com/guestisp/da32982c752850f81b3137ce74aedc68

supermerill, 2020-05-12:
can i use your script to create the appimage directly in the build action?

Do you need to have a name without the date?

guestisp, 2020-05-12:
Sure, just use it if you like. A date is ok.

c1olli, 2020-06-02:
@guestisp 
With the new version called superslicer your scipt is not working properly anymore. Can you update your script?

c1olli, 2020-06-03:
Perfect - I used that tool for the previous version and it worked properly.

guestisp, 2020-06-05:
@c1olli 
> With the new version called superslicer your scipt is not working properly anymore. Can you update your script?

sure, meanwhile could you please tell me which error are you getting ?

c1olli, 2020-06-05:
It says always file or folder not found. I think it is because of the change of names from Slicer++ to SuperSlicer

Errors (sorry they are in german)

build_appimage.sh: Zeile 29: cd: /tmp/Slic3r++_NxMb0ir2xk/Slic3r-2.2.50: Datei oder Verzeichnis nicht gefunden
CMake Error: The source directory "/tmp/Slic3r++_NxMb0ir2xk/deps" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
make: *** Es wurden keine Ziele angegeben und keine „make“-Steuerdatei gefunden.  Schluss.
build_appimage.sh: Zeile 35: cd: /tmp/Slic3r++_NxMb0ir2xk/Slic3r-2.2.50: Datei oder Verzeichnis nicht gefunden
CMake Error: The source directory "/tmp/Slic3r++_NxMb0ir2xk/deps/build" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
make: *** Keine Regel, um „slic3r“ zu erstellen.  Schluss.
cp: Aufruf von stat für '../resources' nicht möglich: Datei oder Verzeichnis nicht gefunden
cp: Aufruf von stat für 'src/slic3r++' nicht möglich: Datei oder Verzeichnis nicht gefunden
--2020-06-05 12:22:28--  https://github.com/AppImage/AppImageKit/releases/latest/download/appimagetool-x86_64.AppImage
Auflösen des Hostnamens github.com (github.com) … 140.82.118.3
Verbindungsaufbau zu github.com (github.com)|140.82.118.3|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 302 Found
Platz: https://github.com/AppImage/AppImageKit/releases/download/12/appimagetool-x86_64.AppImage [folgend]
--2020-06-05 12:22:30--  https://github.com/AppImage/AppImageKit/releases/download/12/appimagetool-x86_64.AppImage
Wiederverwendung der bestehenden Verbindung zu github.com:443.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 302 Found
Platz: https://github-production-release-asset-2e65be.s3.amazonaws.com/9435153/b74f7000-6c54-11e9-90c2-1f11aaf02f75?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20200605%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20200605T102230Z&X-Amz-Expires=300&X-Amz-Signature=c9c2ab0e8280d1ab768781eabe1c303a42a1bd372e1f89ad28efc9a55a0c122e&X-Amz-SignedHeaders=host&actor_id=0&repo_id=9435153&response-content-disposition=attachment%3B%20filename%3Dappimagetool-x86_64.AppImage&response-content-type=application%2Foctet-stream [folgend]
--2020-06-05 12:22:30--  https://github-production-release-asset-2e65be.s3.amazonaws.com/9435153/b74f7000-6c54-11e9-90c2-1f11aaf02f75?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20200605%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20200605T102230Z&X-Amz-Expires=300&X-Amz-Signature=c9c2ab0e8280d1ab768781eabe1c303a42a1bd372e1f89ad28efc9a55a0c122e&X-Amz-SignedHeaders=host&actor_id=0&repo_id=9435153&response-content-disposition=attachment%3B%20filename%3Dappimagetool-x86_64.AppImage&response-content-type=application%2Foctet-stream
Auflösen des Hostnamens github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com) … 52.216.115.19
Verbindungsaufbau zu github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)|52.216.115.19|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 1974248 (1,9M) [application/octet-stream]
Wird in »/tmp/Slic3r++_NxMb0ir2xk/appimagetool.AppImage« gespeichert.

/tmp/Slic3r++_NxMb0 100%[===================>]   1,88M  2,70MB/s    in 0,7s    

2020-06-05 12:22:31 (2,70 MB/s) - »/tmp/Slic3r++_NxMb0ir2xk/appimagetool.AppImage« gespeichert [1974248/1974248]

sed: bin/slic3r++ kann nicht gelesen werden: Datei oder Verzeichnis nicht gefunden
cp: Aufruf von stat für 'resources/icons/Slic3r_192px.png' nicht möglich: Datei oder Verzeichnis nicht gefunden
appimagetool, continuous build (commit effcebc), build 2084 built on 2019-05-01 21:02:41 UTC
Unable to guess the architecture of the AppDir source directory "/tmp/Slic3r++_NxMb0ir2xk/Slic3r-2.2.50/Slic3r++.AppDir/."
A valid architecture with the ARCH environmental variable should be provided
e.g. ARCH=x86_64 appimagetool ...
mv: Aufruf von stat für '/tmp/Slic3r++_NxMb0ir2xk/Slic3r++-*.AppImage' nicht möglich: Datei oder Verzeichnis nicht gefunden

probonopd, 2020-06-22:
Here is a way to produce an AppImage that works for me:

```
DLD=$(wget -q "https://github.com/supermerill/SuperSlicer/releases" -O - | grep ".tar.zip" | grep "_linux_" | grep "/supermerill/" | cut -d '"' -f 2 | head -n 1)
wget -c "https://github.com/$DLD"
export VERSION=$(ls SuperSlicer*.tar.zip | cut -d _ -f 2)
unzip SuperSlicer_*.tar.zip
mkdir -p appdir/usr/share/applications/
( cd appdir/usr ; tar xfv ../../SuperSlicer.tar )
cat > appdir/usr/share/applications/superslicer.desktop <<\EOF
[Desktop Entry]
Name=SuperSlicer
Exec=superslicer
Icon=superslicer
Terminal=false
Type=Application
MimeType=model/stl;application/vnd.ms-3mfdocument;application/prs.wavefront-obj;application/x-amf;
Categories=Graphics;3DGraphics;
Keywords=3D;Printing;Slicer;
EOF
mkdir -p appdir/usr/share/icons/hicolor/128x128/apps
cp appdir/usr/resources/icons/Slic3r_128px.png appdir/usr/share/icons/hicolor/128x128/apps/superslicer.png
cp appdir/usr/resources/icons/Slic3r_128px.png appdir/superslicer.png
wget -c https://github.com/$(wget -q https://github.com/probonopd/go-appimage/releases -O - | grep "appimagetool-.*-x86_64.AppImage" | head -n 1 | cut -d '"' -f 2)
chmod +x appimagetool-*.AppImage
mkdir -p appdir/usr/lib/gtk-2.0/ # FIXME: Let appimagetool do this
cp -r /usr/lib/x86_64-linux-gnu/gtk-2.0/*/* appdir/usr/lib/gtk-2.0/ # FIXME: Let appimagetool do this
./appimagetool-*.AppImage deploy appdir/usr/share/applications/*.desktop
./appimagetool-*.AppImage ./appdir/
```

Of course the first two lines would not be needed in case this gets run as part of the existing build pipeline.

supermerill, 2020-07-03:
I've created one. the script use is on CI branch.

c1olli, 2020-07-04:
I used the scipt from @probonopd without the first 2 lines and changed the zip to tar.zip, then it worked perfekt.
But alway a appimage for download is the nicer solution ;)
I downloaded also the new appimage from your github and it works properly.
Thanks for that.

probonopd, 2020-07-04:
It's great to see an AppImage. However, if currently fails to run on Ubuntu xenial:

```
/run/firejail/appimage/.appimage-9158/bin/superslicer: /lib/x86_64-linux-gnu/libz.so.1: version `ZLIB_1.2.9' not found (required by /run/firejail/appimage/.appimage-9158/bin/superslicer)

/run/firejail/appimage/.appimage-9158/bin/superslicer: /lib/x86_64-linux-gnu/libgcc_s.so.1: version `GCC_7.0.0' not found (required by /run/firejail/appimage/.appimage-9158/bin/superslicer)

/run/firejail/appimage/.appimage-9158/bin/superslicer: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.27' not found (required by /run/firejail/appimage/.appimage-9158/bin/superslicer)

/run/firejail/appimage/.appimage-9158/bin/superslicer: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /run/firejail/appimage/.appimage-9158/bin/superslicer)

/run/firejail/appimage/.appimage-9158/bin/superslicer: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `CXXABI_1.3.11' not found (required by /run/firejail/appimage/.appimage-9158/bin/superslicer)

/run/firejail/appimage/.appimage-9158/bin/superslicer: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /run/firejail/appimage/.appimage-9158/bin/superslicer)

/run/firejail/appimage/.appimage-9158/bin/superslicer: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.22' not found (required by /run/firejail/appimage/.appimage-9158/bin/superslicer)

/run/firejail/appimage/.appimage-9158/bin/superslicer: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.27' not found (required by /run/firejail/appimage/.appimage-9158/bin/superslicer)

/run/firejail/appimage/.appimage-9158/bin/superslicer: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /run/firejail/appimage/.appimage-9158/bin/superslicer)
```

For practical reasons, we recommend to build on the oldest still-supported Ubuntu release, which is xenial at this point.

Please see https://github.com/AppImage/AppImageKit/wiki/Creating-AppImages/cc2441518975caca23e9ce2dba6f08a22c678d1e#creating-appimages-that-are-compatible-with-many-systems.

c1olli, 2020-07-04:
I have the 20.04 LTS Version - the it runs properly.

probonopd, 2020-07-04:
Sure. I am talking about 16.04, which is currently the oldest still-supported Ubuntu release.

supermerill, 2020-07-04:
erf, i built it on the latest, maybe that's why.
I'll create an other vm on 16.04

supermerill, 2020-07-09:
Impossible to compile on 16.04, seems like the installed g++ isn't new enough.

I'll now compile on 18.04. So It seems I won't be compliant until April 2021.

probonopd, 2020-07-23:
Strange thing is that Slic3r and PrusaSlicer somehow seem to be able to compile in a way that can run on 16.04?

supermerill, 2020-07-23:
that's because i added things that need a higher version of wxwidget, and then it escalates.

probonopd, 2020-07-30:
To me, it'd definitely seem worthwhile to support all still-supported Linux distributions.
It's also a requirement in order to be listed on https://appimage.github.io/.

supermerill, 2020-07-31:
If someone can make the latest version compiles (statically) with wxwidget 3.1 on 16.04, i'm ok to use that work to change my scripts.

closed 2020-08-24 by supermerill