supermerill/SuperSlicer #226: Wipe on external perimeters

troypouliot, 2020-05-04:
### Version
2.2.49

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Voron - Klipper

_Is this a new feature request?_ YES

I have not seen this feature in any of the Slic3r branches. 
It would be beneficial for seam locations if you could force a wipe at the end of any external perimeters, regardless of having a retraction or not. It helps blend the stop into the start.

To me, the current wipe option is pretty useless since it only wipes at retractions, not at the end of loops.

Simplify3D has this option and I found it helpful when I was using that slicer.
[image]

Now that I think of it, the default behavior in Simplify (if you enable wipe) is to wipe at the end of EVERY loop. There is another checkbox that tells it to only wipe on external perimeters.
[image]

supermerill, 2020-05-04:
Slic3r wipe by default on every external loop end. Maybe you want to be able to tune it?
The problem is that it's wiping by going a bit inside the part, and if you go too far, you'll going to traverse it entirely.

troypouliot, 2020-05-04:
Do you know what the Prusaslicer behavior is? I can not get it to wipe on loop ends, only on retracts.
Sorry, I am assuming that Slic3r and Prusaslicer behave the same in the wipe aspect.

supermerill, 2020-05-04:
should be the same. slice a cube, look at generated gcode:

``
; printing object Shape-Box id:0 copy 0
[...]
G1 X111.720 Y78.823 E0.92967
G1 X111.720 Y97.146 E0.92967
G1 X93.397 Y97.146 E0.92967
G1 X93.397 Y78.883 E0.92663
G1 X92.888 Y78.315 F10800.000 ; GO TO EXTERNAL LINE
G1 F1800.000
G1 X112.228 Y78.315 E0.98127
G1 X112.228 Y97.655 E0.98127
G1 X92.888 Y97.655 E0.98127
G1 X92.888 Y78.375 E0.97823
G1 X93.275 Y78.418 F10800.000 ; WIPE
G1 E-6.00000 F3600.00000
``

troypouliot, 2020-05-04:
Is there something other than the "wipe while retracting" checkbox to enable wiping elsewhere?

supermerill, 2020-05-04:
no, where so you want to wipe and why?

troypouliot, 2020-05-04:
Green line represents the layer seam. I'd like the last loop (external perimeter) to stop extruding at the green line, but continue traveling along path of the exterior for x.xx amount.

The parts I make need to be acetone smoothed, so minimizing seams is critical. I have found that setting the seam to rear so that the seam remains one continuous line produces the best results. I'm just looking to improve any way I can. I came from Simplify and the wiping on the exterior loops was the norm, so I can't figure out why I can't get the same thing to happen here.

[image]

supermerill, 2020-05-05:
it's already doing that, but going a bit in the interior.
I can create a setting to allow for a greater wipe...

troypouliot, 2020-05-05:
Ahhh, I see it now. It is such a tiny move though, maybe .5mm?. It would be interesting to be able to increase this value if you could. 
If the direction could remain on the current loops' path, then you would not need to worry about using large values since it would continue to follow something that is already there.
It looks like this is what the "wipe on retracts" is doing, except it reverses direction when doing it's wipe. That value is already adjustable, just looking to get that same behavior (except don't reverse direction)
Thanks for all your hard work!

supermerill, 2020-05-11:
the nigthly build https://github.com/supermerill/Slic3r/actions should be build next day.
Can you print with and without, with various form (cube vs cylinder)  and report?
Also, what's the impact if you set the move speed to external perimeter speed (to see if i have to let it at move speed or not)?

liftbag, 2020-05-12:
I think that if the request will be confirmed, the extra wipe option should be added in the filament overrides.

troypouliot, 2020-05-13:
> the nigthly build https://github.com/supermerill/Slic3r/actions should be build next day.
> Can you print with and without, with various form (cube vs cylinder) and report?
> Also, what's the impact if you set the move speed to external perimeter speed (to see if i have to let it at move speed or not)?

I should be able to do some testing tomorrow. I had to work from the actual office last couple days.

closed 2020-06-24 by supermerill