supermerill/SuperSlicer #230: [Request] Add more options to the Height range modifier

th33xitus, 2020-05-09:
I already made a request in the PrusaSlicer repo but it seems no one really is up for such features there. So maybe there is some interest in implementing my suggestions (or at least start a discussion about if they are good/bad) at this place?

https://github.com/prusa3d/PrusaSlicer/issues/3635

### Suggestion
At the moment there are some quite nice options available to choose from in the height range modifier.
But some additional parameters could be quite handy or usefull imho.

**Layers and perimeters:**
- Layer height
- Spiral vase
- Extra perimeters if needed
- Ensure vertical shell thickness
- Avoid crossing perimeters
- Detect thin walls
- Seam position
- External perimeters first

**Infill:** 
- Fill angle
- Infill before perimeters 

**Advanced:**
All Extrusion width related entrys should be listed under the Advanced Settings like in the "Print Settings" tab. At the moment there are some extrusion width settings within an own category.
- Default extrusion width
- Perimeters
- External Perimeters
- Infill
- Solid Infill
- Top solid infill

**Filament:**
- Extrusion multiplier
- Extruder temperature

**Cooling:** (maybe)
- Fan speed

**Extruder:**
- Retraction length
- Lift Z

I know it is quite a lot but it would make the height range modifier an extremely powerfull tool. 
You could actually almost print two objects on the same build plate with a complete different set of settings in one go. It could be (at least kind of) compared to Simplify 3D's so called processes you can set up, which let you choose from the full spectrum of settings S3D has to offer and apply them to different parts of your print (depending on z, so exactly like the height range modifier here) or assign those processes to different objects to your build plate if you have more than one object on there and go for sequential printing for example.

If there are any question why i think option x makes sense to be included there, just ask and i'll try to explain with an example.

supermerill, 2020-05-09:
- Layer height : you should use the variable layer height tool instead (or ask to improve it)
- Spiral vase: spiral vase is a hack and it's impossible to do that without rewriting it entirely
- Extra perimeters if needed: already here, there is a problem ??? or maybe i really don't understand
- Infill : infill what? infill density is already here....
- Solid Infill : already here
- Top solid infill : already here

- Filament : there are nothing in place to override filament settings (so it may be hard/long do implement). and also i don't see why you may want do to that. The temperature influence the color?

- Extruder: same as filament. If you want multiple extruder, just add another, it will do the trick.

th33xitus, 2020-05-09:
Everything below "Suggestion" was meant like the structuring of the height range modifier menu with either already existing options and some options i added to the list by myself (be aware that this was usually meant for PrusaSlicer, i see you already implemented a lot more options into that menu). So it's not meant to be a list of missing options, rather a summary of options which would be nice to have **in total**. Just to clarify that because you wondered about the "extra perimeter if needed" function. There is no problem with it, you just misunderstood or i simply failed to explain it good enough in the first place. sorry.

> Infill : infill what? infill density is already here....

Under the headline "Advanced" i was talking about all **extrusion related** settings. Therefore i didn't meant the infill pattern or infill density, instead i meant the infill extrusion width :) 

> Filament : there are nothing in place to override filament settings (so it may be hard/long do implement). and also i don't see why you may want do to that. The temperature influence the color?

Influecing of the color by the temperature wasn't actually my main point here, rather the possibility to easily create a temperature tower without fiddling around with layer-change-gcode and then save the model as a .3mf project with all those layer dependent temperature changes instead. :)

> Extruder: same as filament. If you want multiple extruder, just add another, it will do the trick.

I didn't meant to use multiple extruders, i meant to be able to edit the retraction length and the z-hop depending on the height of the model. But i guess you are right, one could simply add another extruder, copy all settings and just modify the retraction lengths or z-hop in there. then make a "color change" and apply it that way. The way i saw it was just a way without setting up the printer profile with multiple extruders and just edit those parameters with the height range modifier :)

Well, i hope i cleared up most of the things i tried to communicate? 

I simply see that height range modifier as a really mighty toolset if you have almost all options you have in the normal profiles and be able to make them height dependent.

supermerill, 2020-05-09:
if you have a specific use-case and something isn't available, maybe it's possible to move it, but most of settings that could be available are already here.

There are different level for print settings:
print (it will work for all plater)
object
region
g-code

and if a setting is needed to create the region, it can't be inside!

there are a list of extruders
an extruder "contains" his filament settings.
some features can have their extruder id and it they can be set at a region level.

If you have a specific request, like #186, then you should create an issue for that.
For the retraction thing, there is already an issue about it  #137. 
I can't do everything, as i do that in my free time, so i'm prioritizing what i though is more important (and easy to do).

note: the "height range modifier" is just a shortcut to a modifier that is a big flatty cube. I know that modifiers are really useful, i use them very often.

closed 2020-08-25 by supermerill