supermerill/SuperSlicer #231: Add option in help to load calibration "projects"

supermerill, 2020-05-09:
Problem: it's difficult to fine-tune a slicer for a printer, even more if the printer isn't well tuned, or the filament is bad.

solution: 
 - add buttons to create geometry & modifiers tuned for the loaded printer profile.
 - add a help html page to explain the actions to do when looking at the printed result.

-> bed leveling
-> extrusion volume (extruder and/or filament)
-> temperature / cooling (filament)

others things?

DeviousPenguin, 2020-05-09:
> others things?

I would say speed + acceleration also

supermerill, 2020-06-03:
TODO:
 - test for bridge in petg (as it's too easy for pla)
 - retraction -> ooze test (2 pillars)
- speed -> ringing

closed 2020-10-01 by supermerill