supermerill/SuperSlicer #232: Do not auto-generate G-code when using Klipper

c10l, 2020-05-10:
### Version
2.2.49

### Operating system type + version
macOS Catalina (probably unrelated)

### 3D printer brand / version + firmware version (if known)
Klipper firmware `v0.8.0-520-g787ed452` (although this is unrelatad since this is a feature request for generated G-Code)

### Behavior
I use Klipper firmware and have a `START_PRINT` macro that does the entire start procedure, including heating up bed and nozzle in a specific pattern.

If my `Start G-code` in the `Printer Settings -> Custom G-code` doesn't explicitly set the bed temperature, Slic3r++ prepends my macro with:
```
M107
M190 S<bed_temp>
M104 S<extruder_temp>
```

That bit of G-code interferes with my startup sequence.

* _Expected Results_
First lines of G-code:
```
START_PRINT EXTRUDER=230 BED=70
```

* _Actual Results_
First lines of G-code:
```
M107
M190 S70 ; set bed temperature and wait for it to be reached
M104 S230 ; set temperature
START_PRINT EXTRUDER=230 BED=70
```

_Is this a new feature request?_
Yes, I'd like the option to suppress this automatic G-code generation. It could be a configuration flag that gets enabled when I select `Klipper` as the G-code flavour.

#### Project File (.3MF) where problem occurs
[project.3mf.gz](https://github.com/supermerill/Slic3r/files/4605819/project.3mf.gz)

supermerill, 2020-05-10:
Will do, when i'll change the scripting engine, hooks & gui.

Personally, i don't see the advantage to write the macro in klipper instead of writing it in the slicer.

c10l, 2020-05-10:
Thanks! I love macros because I can keep things in Klipper and don't need to reslice if I change them.
I have a relatievly fiddly hotend+probe and I'm trying a few different song and dances to get it to set the first layer right the first try, and once I've got it I'd rather keep it in one place.
It's personal preference I guess. :)

Anyway, thanks for considering this. Your fork is awesome!

DeviousPenguin, 2020-05-10:
I may have a solution for this as I'm doing something similar with Klipper, here;s my gcode in the slicer:

```gc
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
START_PRINT ; Klipper Start GCode Macro
```

And here's where my macro gets the temps in my klipper macro:

```gc
[gcode_macro START_PRINT]
#default_parameter_BED_TEMP: 60
#default_parameter_EXTRUDER_TEMP: 190
gcode:
    CLEAR_PAUSE                     ; Clears the Pause state
    SET_IDLE_TIMEOUT TIMEOUT=600    ; set the idle timeout (in seconds)
    # Wait for bed to reach temperature
    M190 S{printer.heater_bed.target}
    # Set and wait for nozzle to reach temperature
    M109 S{printer.extruder.target}
    # Temps are good
   ...
```

Note you need to be running an up to date install of Klipper for this to work.

c10l, 2020-05-11:
I fail to see how this is a solution. The point of this issue is to have the whole startup procedure in the macro. In your solution you have `M104` and `M140` in the slicer, running before the macro. That defeats the whole purpose.

DeviousPenguin, 2020-05-11:
Ok maybe what I posted isn't really a solution, more of a workaround, it's just what I use so I can set LED's to change colour while warming up the bed and nozzle for example.

supermerill, 2020-05-11:
builds without these gcode are available here.
https://github.com/supermerill/Slic3r/actions

c10l, 2020-05-11:
That's awesome, very quick turnaround! :)

I've just tested it, looks great!

closed 2020-06-24 by supermerill

ChasonDeshotel, 2022-04-22:
> Personally, i don't see the advantage to write the macro in klipper instead of writing it in the slicer. 

Config files are always better. In this case:

- keeps the printer slicer-agnostic
- network share config file between printers
- don't have to keep laptop and desktop slicers in sync (I tweak from presets)
- track changes with git
- use a real text editor 

> It's harder to edit.

Any text that I can't edit with Vim is harder to edit ;)

`ssh pi` then `kc` (alias for `vim ~/klipper_config/klipper.conf`). Then when the file is open I can`/` search for macros and edit all of them without ever leaving the keyboard

I can comment blocks of code at once, *have an undo buffer*, get syntax highlighting, etc, etc...

Anyway I found the checkbox 😊