supermerill/SuperSlicer #233: No preferences/presets/filaments are saved after restart

manu7irl, 2020-05-11:
### Version
_Version of Slic3r++ used goes here_
VORON EDITION
_Use `About->About Slic3r++` for release versions_
2.2.50
_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
WINDOWS 10 PRO
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
TEVO LITTLE MONSTER + KLIPPER v0.8
### Behavior
* _Describe the problem_
no presets are loaded after the wizard completed, also modifying preferences and restarting the gui does not save any preferences.
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r++*__ preview are preferred_
[image]
[image]
[image]

_Is this a new feature request?_
no
#### Project File (.3MF) where problem occurs
_Upload a Slic3r++ Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r++, Slic3r PE 1.42.0-alpha and newer)_

supermerill, 2020-05-11:
what did you use in the wizard? Custom printer? how can i reproduce?

manu7irl, 2020-05-11:
yeah I went straight to custom and add my delta named TLM

‫בתאריך יום ב׳, 11 במאי 2020 ב-14:06 מאת ‪Merill‬‏ <‪
notifications@github.com‬‏>:‬

supermerill, 2020-05-11:
when i start, i am here 
[image]
then i create a custom printer
[image]
then i select marlin
then i set the bed as round
[image]
then i click on next for dimension and temperature, then i click finish
and i am here
[image]

So i don't see how or what is your problem.

manu7irl, 2020-05-11:
the delta stays, but no presets nor filaments types listed!

neophyl, 2020-05-11:
There won’t be any presets or filaments listed as all the presets for both filaments and print settings belong to the vendor spaces, ie Prusa or creality. That was a breaking change made in Prusa slicer 2.2.0.

To get them you have to create copies with the inherits detached.  If you add a Prusa printer you will get them when that printer is selected.  You can then select the preset and I think it’s under advanced but there’s a detach preset button. Once detached you can then save a copy and it will be available to all printers as it’s no longer part of any vendor space. 

I do have a zip file I put together with all the filaments copied with the inherits removed. It can be imported as a config bundle which would then add all the filaments at least. If you want it lmk and I will attach it.

manu7irl, 2020-05-11:
Wow this is amazing news.
It will be awesome.
And btw the work here is remarkable!
Thanks for your help so much!

neophyl, 2020-05-11:
Here's the zip file containing the filament presets.  Extract and then in Slicer go to File> Import> Import Config Bundle.
[GenericFilaments_Neo_20200401.zip](https://github.com/supermerill/Slic3r/files/4611149/GenericFilaments_Neo_20200401.zip)

manu7irl, 2020-05-11:
Thank you so much

supermerill, 2021-03-19:
should be ok now, right?