supermerill/SuperSlicer #234: Unpredictable behavior in automatic bridges

liftbag, 2020-05-11:
### Version
_2.2.50 (also in PrusaSlicer)_
### Operating system type + version
_macOS Catalina 10.15.4_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
### Behavior
* _Sorry to report it here, as this is an error imported from PrusaSlicer. If you deem it appropriate, I also report it in PrusaSlicer.
Automatically generated bridges have two types of inaccuracies, which depends on the angle of the part_

This is an object created to evaluate the automatic generation of bridges. As can be seen, the most accurate bridge is the one below. The rest show parallelism defects with the general trend of the perimeters, and a detachment from the perimeters, due to the starting point.

[image]

The strategies adopted vary according to the rotation angle of the object: in this case, all bridges do not run parallel to the perimeters

[image]

This is how it would be correct to make a bridge

[image]

Here we see the lack of parallelism between the bridges and the perimeters

[image]

This is probably difficult to solve, as the starting point of the bridge filling is determined by the general slicing algorithm.

[image]

Simplify3D also suffers from the detachment of the bridges from the perimeters, but the angle is correct instead.

[image]

#### Project File (.3MF) where problem occurs
[bridge_angle_test_five_arms.3mf.zip](https://github.com/supermerill/Slic3r/files/4610038/bridge_angle_test_five_arms.3mf.zip)

supermerill, 2020-05-11:
yes, it's not optimal and a bit difficult to improve.

It doesn't known where the periemters are. It has an area to fill and an area where it can anchor the bridge. With that, it tries many angle and keep the one where there are the "most area filled" with anchor. Maybe that's why sometimes it's not strait, to try to fill more area... have to add some test cases.
Problem with the gap, is that if it doesn't go just at the right width, the current algorithm isn't smart enough to left a gap in the middle and fill it with a little bridge/gapfill.

liftbag, 2020-05-11:
> It's a big project to rewrite it.

I actually imagined it.

closed 2021-11-19 by supermerill