supermerill/SuperSlicer #237: Feature Request: Jerk Setting for Perimter / Infill / First Layer...

Argolein, 2020-05-12:
This isn't really a new idea as it is a feature in Cura and Ideamaker.

A good position for movement specific jerk values would be in the speed tab next to the acceleration values for Perimeter, Infill etc.

Would also be nice to respect the G-Code flavor. Marlin uses mm/s for jerk, RepRap mm/min. 

**Whats the benefit?**
E.g. printing perimeter with 5 mm/s (or 300mm/min) jerk, Infill with 8mm/s jerk, Travel 10 mm/s jerk saves print time. 

Picture is from Ideamaker as an example. 

[image]

supermerill, 2020-05-12:
M566 for reprapfirmware
M207 for repetier
other firmware: no possible?

guestisp, 2020-05-12:
M205 on Marlin or Lerdge

guestisp, 2020-05-12:
But how does this work? Before printing a perimeter, the jerk is changed on the fly?

Also, keep in mind that on Marlin, jerk is deprecated, there is a (better) junction deviation speed and it does not differentiate between x, y or Z

@supermerill what do you think in adding support for junction deviation on Marlin firmware? (M205 J) this could differentiate Marlin flavor and Lerdge flavor a Little bit. I can do a pull request if good for you

supermerill, 2020-05-12:
i set, it will be set at each change of type of extrusion.

> what do you think in adding support for junction deviation on Marlin firmware?

what does this mean? what feature I have to wrote?

guestisp, 2020-05-12:
I'll wrote on my own and submit to you

Argolein, 2020-05-12:
I do not have a Marlin 2.0 printer but there is a interesting thread about Junction Deviation it in the Duet forums (RepRap FW):

https://forum.duet3d.com/topic/2629/s-curve-sinusoidal-jerk-acceleration/60?_=1589309512414
(see postings from dc42)

I'm just a newb in coding but if you intend to add the feature, the logic behind it should be easily(?) altered for the corresponding firmware as the firmware (version) only changes the gcode command? The position and such in the gcode file stay the same.

supermerill, 2020-05-12:
well, i have to add all these option, add all triggers (and create the way they work) to hide them, modify the gcode writer 'quite a bit' to add these newgcode at the right place, and in a way that is different for every firmware ....
Not quick to do.

guestisp, 2020-05-12:
All of this for something already deprecated and dismissed in Marlin and other firmwares too.

It's like adding feature to support an already outdated firmware. Marlin 2 is the current stable and use junction speed as default

Argolein, 2020-05-13:
Still you may want to use different JD settings for inner perimeter, Infill, Solid Fill and so on to save print time.

guestisp, 2020-05-13:
I don't think it's the same. JD works differently than jerk and usually you can print faster

Argolein, 2020-05-13:
The calculations are different but the results are quite similar:  lower cornering speed to reduce ringing / ghosting which is wanted for external perimeters.
https://reprap.org/forum/read.php?1,739819

So if you could set a low ringing/ghosting value for external perimeters and a higher value for Infill this would also shave off print time especially when printing Infill like Gyroid.

**As for time difference I added a picture with time estimation from Ideamaker.**

I can't provide you with time estimations for Junction Deviation though because I don't know a Slicer wich knows this feature yet. Maybe Prusa will add it in the future when they add Marlin 2 to their 32 bit printer (Prusa Mini so far) but seeing how long it took them to add LA 1.5. I wouldn't count on it before 2021. 

[image]

albert2004, 2020-05-13:
If this is an request to put extra code before each (or some of) of extrusion types, then it is already possible, as solved in issue #108.

Argolein, 2020-05-13:
Thanks for pointing this out!

I used all available extrusion roles to specify jerk. 

{if extrusion_role=~/.*ExternalPerimeter.*/}M566 X180.00 Y180.00 Z24.00 E300.00               
{elsif extrusion_role=~/.*Perimeter.*/}M566 X360.00 Y360.00 Z24.00 E300.00  
{elsif extrusion_role=~/.*OverhangPerimeter.*/}M566 X360.00 Y360.00 Z24.00 E300.00
{elsif extrusion_role=~/.*InternalInfill.*/}M566 X480.00 Y480.00 Z24.00 E300.00
{elsif extrusion_role=~/.*SolidInfill.*/}M566 X420.00 Y420.00 Z24.00 E300.00
{elsif extrusion_role=~/.*TopSolidInfill.*/}M566 X360.00 Y360.00 Z24.00 E300.00
{elsif extrusion_role=~/.*BridgeInfill.*/}M566 X360.00 Y360.00 Z24.00 E300.00
{elsif extrusion_role=~/.*GapFill.*/}M566 X330.00 Y330.00 Z24.00 E300.00
{elsif extrusion_role=~/.*Skirt.*/}M566 X360.00 Y360.00 Z24.00 E300.00
{elsif extrusion_role=~/.*SupportMaterial.*/}M566 X360.00 Y360.00 Z24.00 E300.00
{elsif extrusion_role=~/.*SupportMaterialInterface.*/}M566 X360.00 Y360.00 Z24.00 E300.00
{elsif extrusion_role=~/.*WipeTower*./}M566 X360.00 Y360.00 Z24.00 E300.00
{else} M566 X480.00 Y480.00 Z24.00 E300.00{endif}

To make things complete the role "Travel" would be nice to have :)
Atm it can happen that a travel move move happens after a external perimeter move with it's settings.

Stephan3, 2020-07-20:
+1 for travel role

The only extrusion role thats missing here and covered by else is "Mixed" BTW
I made a smal pp as a workarround to the goal @Argolein is trying to achieve. Its a 10 minutes write up. Did the Job so far.

launch it like:
```
C:\Users\Stephan\AppData\Local\Programs\Python\Python38-32\python.exe C:\Users\Stephan\Desktop\SuperSlicer_2.2.52.1_win64_20200703\pp.py
```

Example for Klipper:
``` Python
import sys

output = []

with open( sys.argv[1] , 'r' ) as fp:  
	for cnt, line in enumerate(fp):
		if "; retract" in line:
			output.append(line)
			output.append("SET_VELOCITY_LIMIT ACCEL=5000  ACCEL_TO_DECEL=5000 SQUARE_CORNER_VELOCITY=10 \n")
		else:
			output.append(line)

#	overwrite original file
with open( sys.argv[1] , 'w') as f:
	for line in output:
		f.write(line)
```