supermerill/SuperSlicer #238: Bridges/Overhangs that aren't understood as such.

spott, 2020-05-12:
### Version
2.2.49

### Operating system type + version
Mac

### Behavior
overhanging or bridging perimeters that aren't labeled as bridges or overhanging.

[image]

[image]

These should be labeled as overhanging at least, or bridges, but they aren't, which leads to poorer print quality as they are printed to quickly or with too little fan.

#### Project File (.3MF) where problem occurs

[voron_design_cube_v6_slicer++.3mf.zip](https://github.com/supermerill/Slic3r/files/4617461/voron_design_cube_v6_slicer%2B%2B.3mf.zip)

supermerill, 2020-05-12:
Decrease the overhang threshold.

For the thin wall, it's even a bit more complicated than that, at it's not assured that it will began to print from the part... more work to do.

spott, 2020-05-13:
Not the stairs, the single path (perimeter) that floats in space.

I’m not sure what should be done with these, as they are unidirectional and small... but I feel at the very least they should be classified as overhangs (probably better as bridges?) to give them a shot at being printed without falling down.

pranas, 2021-04-20:
I was doing tuning with Voron cube and I encountered the same issue with the debossed X there. I can recognize it in the screenshots and it's probably the easiest way to find reproduce the problem. The overhang stars with a thin line so I even tried to disable thin walls but what is the most interesting is that it would not properly recognize overhang even then.

[image]