supermerill/SuperSlicer #239: [Feature request]Z full step/mm setting

liftbag, 2020-05-12:
### Version
_2.2.50+_
### Operating system type + version
_any_
### 3D printer brand / version + firmware version (if known)
_any_

### Request
* _I just noticed the most recent version of Slic3r has an interesting setting, the Z full step/mm setting_
I've always wondered what level of precision can guarantee the management of continuously variable layers, where the Z axis remains for an entire layer in intermediate positions between the full steps, or in the layers where the supports are generated.
I believe that Ranellucci's approach (or jenkins?) to the problem is substantially correct.
I wonder if it was possible to port this optional and experimental setting also on slic3r++.

The version I have reviewed is the slic3r-1.3.1-dev-54a31eed of 2020-01-27, available here:
https://dl.slic3r.org/dev/win/

[image]

_Is this a new feature request?_
Yes

supermerill, 2020-05-12:
I just tough of that last week
#219
but it's a bit complicated to implement it well.

liftbag, 2020-05-12:
> Won't be used for vase mode, obviously.

Of course :)

supermerill, 2020-06-10:
was more work than i though

liftbag, 2020-06-10:
> was more work than i though

Sorry, I'm a noob about programming. I thought it was possible to bring it directly from slic3r.

Czarnyszarf, 2020-06-24:
What parameters should I use when I don't know the Z steps for my printer?

Can I disable it?

supermerill, 2020-06-24:
put something low like the default one.

Czarnyszarf, 2020-06-24:
by the way, setting it as  0 crashes the program :)

supermerill, 2020-06-24:
erf, i didn't thought of it. Of course, there may be a division by 0...

closed 2020-10-22 by supermerill

Photogad, 2021-03-27:
What setting should I use for a stock Prusa Mk3S?

supermerill, 2021-06-03:
As there is a Mesh bed leveling, the z stepper are never at a fulls tep, so if you want to use this field you need to know the microstep.

based on https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/motor-step-angle-and-screw-pitch-for-mk3-1.8-and-.8mm/, 0.0025