supermerill/SuperSlicer #240: [Request]  xy compensation different between inside and outside of the piece

b1k386, 2020-05-12:
### Version
2.2.50+

### Operating system type + version
any

### 3D printer brand / version + firmware version (if known)
any

### Request

I have been using Slic3r for a while, and I always find many new and very useful functions, one of these was the compensation of the holes, I was wondering if it was possible to add a difference in compensation xy, different between the inside and the outside of the piece , at present if compensation, the compensation is total

supermerill, 2020-05-12:
inside = hole but without the concave restriction, we agree?

b1k386, 2020-05-12:
> inside = hole but without the concave restriction, we agree?
> 
> I wonder why the inside is different from the outside.

yes, in practice if it is a round hole it compensates but if I like a cross it does not compensate, the reason is because it happened to me that two pieces that should fit together and are not round have different tolerances between the inside and the outside

b1k386, 2020-06-05:
> inside = hole ma senza la restrizione concava, siamo d'accordo?
> 
> Mi chiedo perché l'interno sia diverso dall'esterno.

it often happens that the designer designs according to the machine he is using and does not follow rules so to match the pieces on other machines you have to compensate, according to your experience it is possible to divide into

- external compensation only
- internal compensation only
- holes only compensation

closed 2020-08-24 by supermerill