supermerill/SuperSlicer #241: Combine XY and Z moves at the start of the print

combolek, 2020-05-12:
### Version
2.2.50

### Operating system type + version
Windows 10 (18363) 64-bit

### 3D printer brand / version + firmware version (if known)

TwoTrees Sapphire-S upgraded with Klipper v0.8.0-526-gf0c394de and SKR 1.4 Turbo

### Behavior
* _Describe the problem_

At the begining of the print the head always first drops to the initial layer height and then does a fast horizontal move to the correct (x, y) location (this is all after the custom start G-code). 

* _Steps needed to reproduce the problem_

Slice the attached project
Observe the G-code at start (after home and custom g-code):

* _Expected Results_

Single 3D move would be safer in case the bed is not perfectly level
```
G1 X85.317 Y77.452 Z0.200 F9000.000
```

* _Actual Results_

First Z drop, then XY move
```
G1 Z0.200 F9000.000
[...]
G1 X85.317 Y77.452 F9000.000
```

_Is this a new feature request?_

Yes (I think?)

#### Project File (.3MF) where problem occurs
[belt_tensioner.zip](https://github.com/supermerill/Slic3r/files/4618301/belt_tensioner.zip)

supermerill, 2020-05-13:
problem is, before going to z= first_layer_height + zlift, we don't know the z of the printer.
Unless we scan the custom start gcode to be sure it's not too low, we suppose it's at 0.
this feature willc reate many special cases on various places with potential side-effects and complicated safeguards... not very keen of it.

but i think it's possible to add a mega-zlift for the first layer. basically if layer_heught == lift "above_z", then double the z_lift. Enough?

combolek, 2020-05-13:
Yeah, I am certainly not asking for parsing custom gcode. But I don't think I understand the issue. Can we just first go to a safe Z eg. 10mm and then combine the initial Z and XY moves into a single XYZ move. The X, Y and Z values can stay exactly the same as original, there is no need to compute or know anything else. What am I missing?

supermerill, 2020-05-13:
problem is that you have to go back from "10mm" and at the right place, and that is a bunch of special cases, as it interacts with lift, offset, and some others things.

combolek, 2020-05-13:
OK, so can you explain the mega-z-lift workaround? I didn't get that either :-( What is z-lift? Can you describe the sequence of moves that would happen?
Thanks

combolek, 2020-05-14:
OK, so I found that Z lift in Slic3r++ is what Cura calls Z hop on retraction. I then realized that maybe you meant this: lower to the first layer height as it is now, then raise Z by some amount and then do the XY move and then lower again. Is this what you meant? If so, I think this would work as long as the Z raise amount is not tied to the "Z lift on retract", because that may have to be set independently (e.g. forced to be 0). It would have to be a separate parameter but maybe a hard coded value would work too.
Am I understanding correctly?

supermerill, 2020-05-14:
yes.
i was thinking something like 2* first layer hieght. so with Z z-lift and 0.2 first layer height, it will go to 0.6. when moving quikly, for all moves on the first layer.

oh, you can already do that right now. just add an other extruder, set the z-lift of the first extruder to 2mm, slice, add an extruer change at the second layer, and that's it! (it's for testing purpose, as it's a bit of an inconvenience, but it's exactly the behavior i propose).

combolek, 2020-05-14:
I think this would work. So with z-lift=0 and first layer=0.2mm this will travel at 0.4mm? I would not even bother with all travel moves, just the initial move to starting XY. In general I do not like to use any z-hop/z-lift. But if you do implement it for all travel moves then yeah, I think some minimum distance is needed.
Thanks

supermerill, 2020-05-15:
Finally, i accepted that i need to tinker with the current lift, much simpler to do. So it's you first idea, more or less.
you can read the commit comment for explanation of what it does.

combolek, 2020-05-16:
Thanks, I downloaded the latest CI build with this change but I am getting strange results. I set
```
retract_lift = 5
retract_lift_above = 1000
```
but I am still seeing this in the g-code:
```
G1 Z0.200 F9000.000
G1 E-2.00000 F2400.00000
G1 X88.492 Y84.838 F9000.000
```
so it still does the initial travel move at the first layer height. Also, later in the g-code I see this:
```
G1 Z0.600 F9000.000
G1 X106.582 Y95.068
G1 Z0.200
```
???
Project and g-code attached
[belt_tensioner.zip](https://github.com/supermerill/Slic3r/files/4637600/belt_tensioner.zip)

supermerill, 2020-05-16:
oh, i added it before each object, but not the skirt ... i'll correct that

closed 2020-06-24 by supermerill