supermerill/SuperSlicer #242: Voron Test print thread outer most line printed first

Ner036, 2020-05-13:
### Version
Slic3r++ 2.2.5.0.0

_Use `About->About Slic3r++` for release versions_

Printing Voron test prints, small bolt and square nut. Bolt (external thread prints entirely fine)
Thread in block will get stringy, the outermost (innermost - smallest diameter) perimeter is printed first so no material is there to hang on to and gets stringy. probably should do inner perimeters before external perimeters.
[Thread_Test_1_x1_Rev1.zip](https://github.com/supermerill/Slic3r/files/4620141/Thread_Test_1_x1_Rev1.zip)

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
VORON 2.2 350mm

supermerill, 2020-05-13:
can't see that in first 3 layers. at what layer?

Ner036, 2020-05-13:
It seems to be about 60% up inside the nut. I tried several times. It did the thread first then backed its way in. Then outwards like normal for the outsideSent from my Samsung Galaxy smartphone.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/supermerill/Slic3r/issues/242#issuecomment-627828313",
"url": "https://github.com/supermerill/Slic3r/issues/242#issuecomment-627828313",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

supermerill, 2020-05-13:
as there are only 2 perimeters at the thinner area (the thread and the external outer one), it seems that it choose to put the other perimeters with the outer external.
So as it prints : inner perimeter, inner external, outer peri, outer ext.
but it doesn't put any peri in inner, all are in outer. so that's why.

future fix: if inner is only the external and it's not the case for the outer ones, print the inner after the outer.

commit f654b980 (2020-06-01): fix #242
closed 2020-06-01 by supermerill